ros/robot_state_publisher #23: no public default constructor.

Karsten1987, 2015-01-23:
is there any reason for not providing any public default constructor? This prevents any declaration of a robot state publisher as a member variable and initialize it on runtime.

jacquelinekay, 2016-01-22:
No idea. I'll implement this for Kinetic.

jacquelinekay, 2016-01-22:
:turtle:

jacquelinekay, 2016-01-24:
hey @Karsten1987, if you're still interested in this feature, I made a potential implementation for the default constructor of `RobotStatePublisher` in #43. I can also add one for `JointStateListener`.

cheers!

clalancette, 2017-07-27:
@Karsten1987 Are you still interested in this issue?  If so, we can probably pretty easily push it over the finish line.  If not, I'll close it out.  Just let me know.

Karsten1987, 2017-07-27:
i like that once a year, we push that issue forward :D 

I am currently don't have any code which uses it, but I still think the proposal is valid.

clalancette, 2017-07-27:
Haha, nice.  Sure, I'll leave it open, and hopefully get to it soon :).

clalancette, 2021-11-12:
I'm going to close this out.  This is unlikely to ever be done for ROS 1, and the whole thing is rewritten in ROS 2.

PR #87 (2020-04-17): Add in a public default constructor
This should fix #23, and obsoletes #43 .  This PR does a number of things:

1.  It pushes the remaining TF dependency as deeply as possible.  To completely get rid of TF would mean we would have to get rid of the `tf_prefix` functionality.
1.  It moves the include of `boost::shared_ptr` over to the one place it is needed.  We basically can't get rid of this use of boost, since it is what is returned in the callback.
1.  It does some cleanup of namespaces.
1.  It switches the package to package format 2.
1.  It removes the deprecated name of `robot_state_publ [... cut]
closed 2021-11-12 by clalancette