ros/robot_state_publisher #24: Node crashes on Indigo

skasperski, 2015-02-07:
Hi, I am using the robot_state_publisher in Indigo on an Ubuntu 14.04, and no matter what I do, it crashes without further notice.

/opt/ros/indigo/lib/robot_state_publisher/robot_state_publisher
Segmentation fault (core dumped)

It's installed as binary package, version is 1.10.4. 
Is there any other info, I could provide?

Best regards,
Sebastian

wjwwood, 2015-02-27:
Hi @skasperski, sorry for the delay, I wasn't watching this repository, I'm not sure who's actually keeping an eye on it these days.

I'm unable to get it to crash if I setup a `roscore` and set a valid `/robot_description` ROS param.

Have you tried doing a `sudo apt-get update && sudo apt-get dist-upgrade`? It might have been an ABI break in a dependency or something like that.

mryellow, 2015-06-17:
ARM? Is it this issue?

http://answers.ros.org/question/209686/turtlebot-apt-get-update-has-generated-an-error-on-minimallaunch/

NikolasE, 2015-07-30:
the robot_description-parameter would be helpful.

skasperski, 2015-07-30:
It seems like I can't attach any files. The description is created from the definition in the p2os_urdf package. So I start it with:

    <launch>
        <include file="$(find p2os_urdf)/launch/upload_pioneer3at.xml"/>
        <node pkg="robot_state_publisher" type="state_publisher" name="robot_state_publisher">
            <param name="publish_frequency" type="double" value="30.0"/>
    	    <param name="tf_prefix" type="string" value=""/>
        </node>
        <node pkg="p2os_urdf" type="p2os_publisher_3at" name="publisher"/>
    </launch>

jonbinney, 2015-09-12:
I just saw a segmentation fault with my own URDF, on ARM also. Will investigate further.

jonbinney, 2015-09-12:
Compiling from source and running in gdb, I see:
```
Program received signal SIGSEGV, Segmentation fault.
0xb6610dc4 in ?? () from /lib/arm-linux-gnueabihf/libpcre.so.3
(gdb) bt
#0  0xb6610dc4 in ?? () from /lib/arm-linux-gnueabihf/libpcre.so.3
#1  0xb6610e62 in ?? () from /lib/arm-linux-gnueabihf/libpcre.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```
Which is pretty scary....

jonbinney, 2015-09-12:
I get the same result with the simplest URDF I couuld come up with:
```
<robot name="foo" xmlns:xacro="http://ros.org/wiki/xacro"></robot>
```

jonbinney, 2015-09-12:
Here's what it is linked against:
```
$$ ldd robot_state_publisher 
        librobot_state_publisher_solver.so => /home/odroid/ws/rsp/devel/lib/librobot_state_publisher_solver.so (0xb6eed000)
        liborocos-kdl.so.1.3 => /opt/ros/indigo/lib/liborocos-kdl.so.1.3 (0xb6e7c000)
        libkdl_parser.so => /opt/ros/indigo/lib/libkdl_parser.so (0xb6e6f000)
        liburdf.so => /opt/ros/indigo/lib/liburdf.so (0xb6e49000)
        libroscpp.so => /opt/ros/indigo/lib/libroscpp.so (0xb6d58000)
        librosconsole.so => /opt/ros/indigo/lib/librosconsole.so (0xb6d32000)
        libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0xb6d27000)
        librostime.so => /opt/ros/indigo/lib/librostime.so (0xb6d01000)
        libboost_system.so.1.54.0 => /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.54.0 (0xb6cce000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6c24000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6c02000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6b1b000)
        libtf_conversions.so => /opt/ros/indigo/lib/libtf_conversions.so (0xb6b10000)
        libtf.so => /opt/ros/indigo/lib/libtf.so (0xb6ae7000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6a7b000)
        /lib/ld-linux-armhf.so.3 (0xb6f0d000)
        libtinyxml.so.2.6.2 => /usr/lib/arm-linux-gnueabihf/libtinyxml.so.2.6.2 (0xb6a64000)
        liburdfdom_world.so.0.2 => /usr/lib/arm-linux-gnueabihf/liburdfdom_world.so.0.2 (0xb6a3f000)
        libclass_loader.so => /opt/ros/indigo/lib/libclass_loader.so (0xb6a21000)
        libroslib.so => /opt/ros/indigo/lib/libroslib.so (0xb6a0d000)
        librosconsole_bridge.so => /opt/ros/indigo/lib/librosconsole_bridge.so (0xb6a02000)
        libboost_filesystem.so.1.54.0 => /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.54.0 (0xb69e9000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb69ce000)
        libconsole_bridge.so.0.2 => /usr/lib/arm-linux-gnueabihf/libconsole_bridge.so.0.2 (0xb69bf000)
        liburdfdom_model.so.0.2 => /usr/lib/arm-linux-gnueabihf/liburdfdom_model.so.0.2 (0xb699b000)
        libpcrecpp.so.0 => /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0 (0xb698d000)
        libxmlrpcpp.so => /opt/ros/indigo/lib/libxmlrpcpp.so (0xb696f000)
        libcpp_common.so => /opt/ros/indigo/lib/libcpp_common.so (0xb6961000)
        libboost_thread.so.1.54.0 => /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.54.0 (0xb6949000)
        librosconsole_log4cxx.so => /opt/ros/indigo/lib/librosconsole_log4cxx.so (0xb6934000)
        librosconsole_backend_interface.so => /opt/ros/indigo/lib/librosconsole_backend_interface.so (0xb692a000)
        liblog4cxx.so.10 => /usr/lib/liblog4cxx.so.10 (0xb67f2000)
        libboost_regex.so.1.54.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.54.0 (0xb6743000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6735000)
        libkdl_conversions.so => /opt/ros/indigo/lib/libkdl_conversions.so (0xb672a000)
        libtf2_ros.so => /opt/ros/indigo/lib/libtf2_ros.so (0xb66ad000)
        libtf2.so => /opt/ros/indigo/lib/libtf2.so (0xb6683000)
        libPocoFoundation.so.9 => /usr/lib/libPocoFoundation.so.9 (0xb65a1000)
        librospack.so => /opt/ros/indigo/lib/librospack.so (0xb656c000)
        libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0xb6535000)
        libaprutil-1.so.0 => /usr/lib/arm-linux-gnueabihf/libaprutil-1.so.0 (0xb6513000)
        libapr-1.so.0 => /usr/lib/arm-linux-gnueabihf/libapr-1.so.0 (0xb64ef000)
        libicuuc.so.52 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.52 (0xb63f1000)
        libicui18n.so.52 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.52 (0xb629c000)
        libactionlib.so => /opt/ros/indigo/lib/libactionlib.so (0xb627f000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb6263000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6258000)
        libboost_program_options.so.1.54.0 => /usr/lib/arm-linux-gnueabihf/libboost_program_options.so.1.54.0 (0xb6204000)
        libpython2.7.so.1.0 => /usr/lib/arm-linux-gnueabihf/libpython2.7.so.1.0 (0xb5fa6000)
        libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0xb5f6f000)
        libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0xb5f4e000)
        libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0xb5f42000)
        libicudata.so.52 => /usr/lib/arm-linux-gnueabihf/libicudata.so.52 (0xb48ce000)
        libutil.so.1 => /lib/arm-linux-gnueabihf/libutil.so.1 (0xb48c3000)
```

jonbinney, 2015-09-12:
Valgrind has lots of scary things to say. Here's a small excerpt:
```
==13972== Use of uninitialised value of size 4
==13972==    at 0x40072F4: _dl_lookup_symbol_x (dl-lookup.c:838)
==13972==    by 0x400A9CD: _dl_fixup (dl-runtime.c:111)
==13972==    by 0x400F19B: _dl_runtime_resolve (dl-trampoline.S:57)
==13972==    by 0x4D62EC7: __pthread_initialize_minimal (nptl-init.c:374)
==13972==    by 0x4D61CC7: ??? (in /lib/arm-linux-gnueabihf/libpthread-2.19.so)
==13972== 
==13972== Conditional jump or move depends on uninitialised value(s)
==13972==    at 0x400A9DA: _dl_fixup (dl-runtime.c:125)
==13972==    by 0x400F19B: _dl_runtime_resolve (dl-trampoline.S:57)
==13972==    by 0x4D62EC7: __pthread_initialize_minimal (nptl-init.c:374)
==13972==    by 0x4D61CC7: ??? (in /lib/arm-linux-gnueabihf/libpthread-2.19.so)
```

tfoote, 2015-09-12:
There have been users documenting that there is an issue with libpcre on armhf: http://answers.ros.org/question/209686/turtlebot-apt-get-update-has-generated-an-error-on-minimallaunch/ see the answer from @NStiurca

jonbinney, 2015-09-12:
Thanks, I'll take a look at that

jonbinney, 2015-09-13:
After looking at the ROS answers, and doing some debugging, I'm reasonably sure that my issue is the same as https://github.com/ros/robot_model/issues/110 and so I'll continue discussion there.

jacquelinekay, 2016-01-28:
I closed ros/robot_model#110 because I deduced that it's a problem with a downstream package and not urdf. Still no confirmation from @skasperski that the original problem was due to ARM and/or the libpcre issue--but I'm going to close since there wasn't enough information originally provided to determine the cause of the issue, and the other issue brought up in this discussion has been resolved. Please advise if this is not the case.

skasperski, 2016-01-29:
Because I don't know. I was using Indigo on my desktop pc with Ubuntu 14.04. I recently switched to Jade (also Ubuntu 14.04), where the problem still exists.

jacquelinekay, 2016-01-29:
Thanks, sounds like your original issue isn't relevant to ARM. I'll reopen this and investigate a problem with the debs in Jade.

Actually, a user on the Baxter users mailing list recently reported a similar problem:

https://groups.google.com/a/rethinkrobotics.com/forum/?utm_medium=email&utm_source=footer#!msg/brr-users/uG0OU8pYfm4/-O0qb_bmDgAJ

Can you try the instructions Ian posted and report back here when you have a chance?

```
$ sudo apt-get update
$ sudo apt-get install ros-indigo-robot-state-publisher
```

If that doesn't work, consider running `$sudo apt-get dist-upgrade`.

sloretz, 2018-02-13:
I'm able to reproduce a crash on `indigo` when running the empty URDF provided by @jonbinney.

```
$ apt show ros-indigo-robot-state-publisher
Package: ros-indigo-robot-state-publisher
Version: 1.11.2-0trusty-20180110-172235-0800
Maintainer: Ioan Sucan <isucan@google.com>
Installed-Size: 428 kB
Depends: libboost-system1.54.0, libc6 (>= 2.14), libgcc1 (>= 1:4.1.1), libstdc++6 (>= 4.1.1), ros-indigo-orocos-kdl (>= 1.3.0), libeigen3-dev, ros-indigo-catkin, ros-indigo-kdl-parser, ros-indigo-rosconsole, ros-indigo-roscpp, ros-indigo-rostime, ros-indigo-sensor-msgs, ros-indigo-tf, ros-indigo-tf-conversions, ros-indigo-tf2-kdl, ros-indigo-tf2-ros
Homepage: http://wiki.ros.org/robot_state_publisher
Priority: extra
Section: misc
Download-Size: 55.3 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
Description: This package allows you to publish the state of a robot to tf.
 Once the state gets published, it is available to all components in the system that also use tf. The package takes the joint angles of the robot as input and publishes the 3D poses of the robot links, using a kinematic tree model of the robot. The package can both be used as a library and as a ROS node. This package has been well tested and the code is stable. No major changes are planned in the near future
```

```xml
<?xml version="1.0"?>
<launch>
  <param name="robot_description" value='&lt;robot name="foo" xmlns:xacro="http://ros.org/wiki/xacro"&gt;&lt;/robot&gt;'/>
  <node pkg="robot_state_publisher" name="robot_state_publisher" type="robot_state_publisher" output="screen"/>
</launch>
```

```
process[robot_state_publisher-2]: started with pid [316]
[ERROR] [1518481302.405483470]: No link elements found in urdf file
[robot_state_publisher-2] process has died [pid 316, exit code -11, cmd /opt/ros/indigo/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/developer/.ros/log/dd4738a0-1053-11e8-81e1-0242ac110005/robot_state_publisher-2.log].
log file: /home/developer/.ros/log/dd4738a0-1053-11e8-81e1-0242ac110005/robot_state_publisher-2*.log
```

However, the node doesn't crash if the URDF has at least one link

```xml
<?xml version="1.0"?>
<launch>
  <param name="robot_description" value='&lt;robot name="foo" xmlns:xacro="http://ros.org/wiki/xacro"&gt;&lt;link name="some_link"&gt;&lt;/link&gt;&lt;/robot&gt;'/>
  <node pkg="robot_state_publisher" name="robot_state_publisher" type="robot_state_publisher" output="screen"/>
</launch>
```

I'm not sure if an empty URDF is the original issue reported by @skasperski. I'm going to close this since I don't have a way to reproduce the original issue, and I've confirmed the latest version is able to run without crashing. I've also opened a separate issue for the crash with an empty URDF (#81).

closed 2018-02-13 by sloretz