ros/robot_state_publisher #25: Missing support for subclassing

peci1, 2015-02-24:
I'd like to customize the behavior of both joint_state_listener and robot_state_publisher, but the interface doesn't allow for that. In joint_state_listener, all the methods are private. And both in joint_state_listener and robot_state_publisher, no method is virtual to support subclassing. Furthermore, there's no way to "plug in" a different class than RobotStatePublisher to the joint_state_listener.

I'd like to ask you to make this package more friendly for subclassing.

If you'd like to know my motivation for subclassing - in our project, we're forced to fake some fixed joints by prismatic joints with zero limits. And I'd like the robot_state_publisher to recognize these fake fixed joints as fixed and to treat them so. And I'd also like to use the /tf_static topic for publishing the transforms for fixed joints.

jacquelinekay, 2016-01-24:
hey @peci1,

it's a bit late, but I can look into adding better support for subclassing in Kinetic.

peci1, 2016-01-25:
@jacquelinekay  +1

closed 2016-05-23 by jacquelinekay

peci1, 2016-05-23:
Thanks ;) Is there a possibility to get it also merged into indigo-devel?

jacquelinekay, 2016-05-23:
I believe it technically isn't an API change and could go into Indigo, but it will definitely break ABI. I will think about it.