ros/robot_state_publisher #29: Reload robot model (e.g. during calibration)

NikolasE, 2015-07-03:
During the calibration process of my robot, I automatically generate a new urdf that is also written to the parameter server. So far, there is no way to load this parameter without restarting the process. 

I propose some ways to allow for a reload:

a) a regular update (e.g. every 5 seconds)
b) update via short msg (std_msgs/Empty) on a robot_state_publisher/reload_robot_model topic
c) update via service so that a problem (malformed robot_description-parameter) can be returned. 

(I'd be happy to implement it if someone else is going to use it (and it will be merged))

DLu, 2015-07-03:
I'd say c makes the most sense.

NikolasE, 2015-07-04:
What do you think about a simple http://wiki.ros.org/std_srvs Trigger?
http://docs.ros.org/api/std_srvs/html/srv/Trigger.html

ipa-mig, 2015-07-06:
:+1: to c) and a :+1: to std_srvs/Trigger.
I could definitiely use this as well, also for calibration process. Always restarting is quite cumbersome.

NikolasE, 2015-07-29:
Code is finished. What do you guys think?

DLu, 2015-07-29:
@isucan is the official maintainer here, so I'll defer for now.

gavanderhoorn, 2015-07-29:
I have no authority here, but a quick glance shows multiple violations of the ROS/Google C++ code style (no 2 spaces indent, missing underscore suffix on member variables, method names not in CamelCase, etc). These do not influence the functionality, but should still be fixed.

A run with [roslint](http://wiki.ros.org/roslint) should show what needs fixing.

isucan, 2015-07-29:
c) makes most sense to me as well, but I don't think this is a good idea in general. Many nodes load the robot_model once, at startup, create whatever internal representation and continue from that point on. Reloading the robot model will lead to an inconsistent system state. If others agree, we should at least document this caveat carefully when describing the service.

@gavanderhoorn makes a good point, the style guide should be followed.

ipa-mig, 2015-07-30:
I'd like to have this possibility. As discussed it helps a lot when calibrating sensors on the robot.
One could also think of having a `reconfigureable_robot_state_publisher` as a new node that provides this service and you can start it when doing calibration.
Or a parameter could be added that would default to `false` and you would have to set it to true to advertise the service in the first place.

gavanderhoorn, 2015-07-30:
Doing this properly is however really more difficult than just adding 'reload urdf support' to this node. As @isucan already wrote, many nodes load and parse `robot_description` once, and never update it. An inconsistent system is imo worse than one where I can't update it without restarting.

NikolasE, 2015-07-30:
I added the parameter proposed by Matthias. If the user does not set the parameter at launch, no service will be available and the node behaves as before. If the user however knows what he is doing, 
he can use the new functionality. ( And even without this parameter, it's hardly possible to trigger the service unintentionally. )

rethink-imcmahon, 2015-08-20:
I think this is an incredibly important issue, but one that needs to be handled carefully, as it will affect (nearly) every ROS robot. 

To give one use-case from Rethink, the Baxter robot's `roscore` and `robot_state_publisher` stays alive for duration that the robot is powered on. The Rethink end effectors (vacuum grippers, electric grippers, pneumatic grippers, etc.) can be physically be hot-swapped on the robot. The entire robot needs to be aware of such a change at runtime, particularly the `robot_state_publisher`.

This led us to internally implement a [mutable_robot_state_publisher](https://github.com/RethinkRobotics/mutable_robot_state_publisher/tree/mutable_state_publisher) capable of dynamically accepting joint and link fragments from nodes on the network, incorporating them into the URDF, and rebuild any relevant kinematic chains (KDL chains for Baxter's arms). We introduce a new parameter called `robot_base_description` which contains the core robot URDF and is treated as modifiable by the `robot_state_publisher`. Also, rather than a Service or Action interface, we use standard Topics with a [URDFConfiguration.msg] (https://github.com/RethinkRobotics/mutable_robot_state_publisher/blob/mutable_state_publisher/msg/URDFConfiguration.msg) to add or remove fragments. Those new fragments are continually published to the topic with unchanging timestamps. Its assumed that when an timestamp is updated, the `robot_description` parameter needs to be rebuilt. Any new node can come up to speed (and stay updated) by listening to the same `/robot/urdf` topic, then building onto the URDF located in `robot_base_description` parameter. 

I did not implement this `mutable_robot_state_publisher` myself, but I did push for open sourcing it because I feel dynamic URDF construction is a problem that plagues many ROS robots. I think it is pretty elegant for any node to come up, report that it has a new fragment of URDF to add, proceed to publish `joint_states` for any of its new non-fixed joints, have the `robot_state_publisher` begin creating the new transforms without missing a beat.

While I believe the `mutable` publisher is an improvement over the `robot_state_publisher`, I don't think our implementation is ideal - I think we should use some combination of ROS Services and ROS Topics to update the URDF and trigger relevant nodes to reload URDFs. I also think the `robot_base_description` may more sense in Baxter's context than it would other more modifiable robots (although you could make `robot_base_description` just your `root` link). And every node that cares about the URDF needing to build URDF changes along with the `robot_state_publisher` seems redundant to me.

The implementation that I described is by no means set in stone. Its just one possible implementation for changing and reloading URDFs, and I very much welcome any comments / recommendations / criticism to make it better. Admittedly, I should have been following this repository much earlier so I could have chimed in here. Thanks to @DLu for pointing me to this thread.

NikolasE, 2015-09-03:
This sounds really interesting, although the possibility to change subtrees wouldn't be needed for the calibration scenario where only some distances change. 

Is someone going to RosCon? I'll be there for Magazino and we could discuss it in a one of the Birds of a Feather-Slots.

jacquelinekay, 2015-10-19:
What's the current status of this discussion? Was there any progress at ROSCon?

I, for one, think this feature would be an improvement for `robot_state_publisher` and am reviewing the work @NikolasE did now.

rethink-imcmahon, 2015-10-19:
Yes, we had a birds of a feather meeting on the topic of a dynamic robot state publisher at ROSCon. It was a very interesting line of discussion, and the general consensus was that we need a generic way to modify URDF's dynamically. Tully put forth that the `robot_description` parameter should probably have been constructed as a latched topic in the first place. Currently most nodes that use the URDF load the `robot_description` at the time they start and then never check to see if the parameter has changed. If these nodes had callbacks that triggered on URDF change, then that would nudge us towards a more dynamic `robot_state_publisher`. The next step was to open up a REP for this change since it affects a large portion of ROS, which I've yet to do.

jacquelinekay, 2016-01-29:
@rethink-imcmahon hey Ian! I'd like to get the ball rolling on this again. I'm not sure if we have enough time to resolve this before Kinetic, but it would be cool if we could. Would it help if I attempted to write up a REP for the change?

pinging @NikolasE, @DLu, @tfoote

rethink-imcmahon, 2016-01-29:
Thank you for pinging me on this Jackie. I apologize for dropping the ball on this issue after ROSCon. Yes, I completely agree that a push to have this resolved before Kinetic would be worthwhile. This functionality would be a huge gain for reconfigurable robots Kinetic. Let's coordinate efforts to put forth a REP for the change.

My most recent release of Baxter (Dec 28) required this functionality onboard for grippers, so I ended up deploying the unofficial [mutable_robot_state_publisher](https://github.com/RethinkRobotics/mutable_robot_state_publisher) onboard the robot due to a lack of bandwidth (and time) to address this properly in the REP setting.

jacquelinekay, 2016-01-29:
I've started a very rough draft of the REP in my fork of ros_infrastructure/rep:

https://github.com/jacquelinekay/rep/blob/reload_robot_model/rep-reload-robot-model.rst

I'll add you as a collaborator on my fork in case you want to push directly to the document (or maybe through a branch on the fork).

I listed Ian and myself as the authors since we're planning on collaborating, but anyone else is welcome to contribute to the REP--just let me know. We should set a deadline for the first draft--is next week too ambitious?

rkeatin3, 2017-06-12:
What is the status of this feature? I'm also interested in doing online calibration.

My apologies if this isn't the right place to ask for an update.

clalancette, 2017-07-27:
Sorry for the late reply.  While this seems to be an issue that people do need, I don't think I am going to have time for it in the near future.  If you are interested in pushing it forward, it might be interesting to resurrect @jacquelinekay 's REP, and go forward from there.

clalancette, 2018-11-19:
Sorry, no.  In order for this to make progress, someone will have to step up to do the work to move the REP forward, then update the code.  I'm just not going to have time to do it.

closed 2020-12-08 by NikolasE

asiemasz, 2022-01-11:
Hi! @NikolasE, did you somehow resolve the problem?