ros/robot_state_publisher #46: Future dating disappears when use_tf_static is false

stwirth, 2016-05-04:
[This commit](https://github.com/ros/robot_state_publisher/commit/727b84871715040f81348750b99a8969b682a836#diff-66aa2aa2284f8c6591676307958098f5L111) removed future dating of fixed transforms in favor of a new parameter `use_tf_static`. The default value for this parameter is false in kinetic and indigo, true in jade (?!?) so the new default now is to not future date transforms anymore.
I'd say the correct implementation would be to future date if use_tf_static is false.

jacquelinekay, 2016-05-11:
Personally I think that future dating for fixed transforms that don't use tf_static is a hack and fixed transforms should simply use tf_static. I am considering taking out backwards compatibility for this in Kinetic. What do you think of this change?

stwirth, 2016-05-12:
The reason I opened this ticket was that tfs suddenly behaved differently after an upgrade as the future dating disappeared and `use_tf_static` defaulted to false.
So I think it is fine to remove future dating if `use_tf_static` defaults to true to preserve the behavior of `waitForTransform` for static transforms.

jacquelinekay, 2016-05-12:
Ok. I'll add future dating back to Jade and Indigo though, to keep backwards compatibility. Thanks for reporting this.

sloretz, 2017-03-21:
Looks like this issue is resolved by #49. Feel free to reopen if that's not the case.

closed 2017-03-21 by sloretz