ros/robot_state_publisher #58: Mandatory dependency on rostest

jvgomez, 2016-08-30:
Hi,

I have noticed that the dependency on rostest is always there. In other packages ([TF example](https://github.com/ros/geometry/blob/indigo-devel/tf/CMakeLists.txt#L51)) rostest is only find-packaged if `CATKIN_ENABLE_TESTING` is true. [Reference issue](https://github.com/ros/rosdistro/issues/3010)

I wonder whether there is a reason for not doing the same in this package.

jvgomez, 2016-11-01:
Ping. I am interested on knowing the answer to know whether I should do a PR with the fix here (to indigo-devel) or to meta-ros with the corresponding patch.

tfoote, 2016-11-01:
Moving the rostest dependency inside the conditional is better. You can then also move it to be a `test_depend` in the package.xml

PR #66 (2016-11-16): Make rostest dependency conditional, depending on CATKIN_ENABLE_TESTING
Closes #58
closed 2016-11-17 by jvgomez