ros/robot_state_publisher #59: Use new urdf SharedPtr typedefs

nxdefiant, 2016-09-03:
Urdf has moved to SharedPtr which results in a compile error:

```
/opt/ros/ros_catkin_ws/src/robot_state_publisher/src/joint_state_listener.cpp: In function 'int main(int, char**)':
/opt/ros/ros_catkin_ws/src/robot_state_publisher/src/joint_state_listener.cpp:164:98: error: conversion from 'std::map<std::__cxx11::basic_string<char>, std::shared_ptr<urdf::Joint> >::iterator {aka std::_Rb_tree_iterator<std::pair<const std::__cxx11::basic_string<char>, std::shared_ptr<urdf::Joint> > >}' to non-scalar type 'std::map<std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::Joint> >::iterator {aka std::_Rb_tree_iterator<std::pair<const std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::Joint> > >}' requested
   for(std::map< std::string, boost::shared_ptr< urdf::Joint > >::iterator i = model.joints_.begin(); i != model.joints_.end(); i++){
```
(3600 lines following this)

Replacing 
`boost::shared_ptr<urdf::JointMimic>`
with
`urdf::JointMimicSharedPtr`

fixes this error.

sloretz, 2017-04-17:
Looks like this depends on the version of https://github.com/ros/urdfdom_headers. Yakkety and Zesty are using [liburdfdom-headers-dev 1.0](http://packages.ubuntu.com/yakkety/liburdfdom-headers-dev) which has moved on to c++11 shared_ptrs. This issue must be fixed for Lunar which [is supporting those platforms](http://www.ros.org/reps/rep-0003.html).

sloretz, 2017-04-17:
Looks likes this was fixed by #60 on Kinetic, and so is not an issue on Lunar. It is still broken on Indigo and Jade, but the platforms this will happen on are way above the supported platforms listed in [REP-3](http://www.ros.org/reps/rep-0003.html).

It would be nice to get this building on Indigo on higher platforms, but unfortunately I don't see a path for that. The [version of liburdfdom-headers-dev in trusty](http://packages.ubuntu.com/trusty/all/liburdfdom-headers-dev/filelist) does not have [urdf_model/types.h](https://github.com/ros/urdfdom_headers/blob/0.4/urdf_model/include/urdf_model/types.h)

I'll close this issue for now. Feel free to reopen if you see a path forward.

closed 2017-04-17 by sloretz