ros/robot_state_publisher #63: Using equations as mimic multipliers

NickHornsey, 2016-09-29:
Hello, 
I am currently working on a urdf and I want to create a mimic joint which moves in relation to the previous joint using the formula y=0.010x^2-0.540x+30.5 where y = the value of the mimic joint and x = the angle of the parent link. Is there a way to do this? 

Thank you

DLu, 2016-09-29:
To my knowledge, that is not possible, it is only possible to do linear mimics.

NickHornsey, 2016-09-30:
Thanks for getting back to me, do you know of any ways to work around this problem?

skohlbr, 2016-10-23:
You could write your own little node that publishes the (slave) mimic joint's state based on the joint state of the master joint. That is probably the quickest solution, if not the most generic one.

sloretz, 2017-03-21:
Hi @NickHornsey. I think http://answers.ros.org/questions/ would be a better place for this. Feel free to reopen.

closed 2017-03-21 by sloretz