ros/robot_state_publisher #67: Breaking change in 1.13.3 with static transforms

zacwitte, 2016-11-15:
It looks like commit e957b26827d1be4866e3e6ef07aec59ad1cf2a2b broke static transforms defined in URDF. Joints of type fixed no longer show up in the TF tree. They're latched, but not published continuously and TF doesn't know about them. The work around is to define the param `use_tf_static` as false on robot_state_publisher, but this breaking behavior should not happen in a minor version bump.

wjwwood, 2016-11-15:
@zacwitte I understand your annoyance, we try hard not to make changes like this in existing distributions, but this was not the intended behavior for Kinetic. As the commit you referenced says, `use_tf_static` defaults to true in Jade and should have continued to do so in Kinetic, but it got changed back to defaulting to false by accident. In that light I believe it was a bug fix for Kinetic.

I apologize for the break in Kinetic, but unless there is an overriding consensus to go back to defaulting to false (as it was in Indigo, but different from what is the case in Jade) I don't think we should change it again.

A separate concern (which might indicate we should keep defaulting to false, despite the Jade behavior) was brought up by @jonbinney, which is that the Python tf interface doesn't seem to support the static transforms correctly.

See: https://github.com/ros/robot_state_publisher/pull/56

@meyerj is my understanding of events correct?

/cc @DLu @isucan

jonbinney, 2016-11-15:
@zacwitte when you say "TF doesn't know about them" are you using C++ or python (or the command line tools?) I ran into this problem with python in particular since it doesn't know about `/tf_static` topic. TF2 should work fine. Personally I think that now that `use_tf_static` defaults to true, tf python needs to be deprecated, with big huge warnings to keep people from using it. Otherwise new people are going to continually running into this and wasting time debugging.

meyerj, 2016-11-26:
> @meyerj is my understanding of events correct?

Yes. I was not aware that Python tf does not support `tf_static` yet. But I would consider this as a bug in tf that will hopefully be fixed soon in https://github.com/ros/geometry/pull/134 thanks to @roehling. The robot_state_publisher is probably not the only source of static transformations, so staying backwards-compatible here would not solve the real problem.

sloretz, 2017-03-21:
It looks like this issue can be closed. Feel free to reopen.

closed 2017-03-21 by sloretz