ros/robot_state_publisher #73: recipe for target 'CMakeFiles/robot_state_publisher_solver.dir/src/robot_state_publisher.cpp.o' failed

khalid2696, 2017-04-06:
I'm trying to install ros-robot on RPi running Raspbian Jessie. When I'm building the catkin space, I'm getting the above mentioned error. All the 117 packages before this installed without any errors.

Any help would be appreciated.

sloretz, 2017-04-06:
Would you mind posting the console output? Possibly a duplicate of #59

khalid2696, 2017-04-06:
Lots of error cropped up. PFA a part of the error log.

Of course, I cloned the recent version of robot_state_publisher. And it has the new version of the pointer - urdf::JointMimicSharedPtr instead of the boost one.

[Error.txt](https://github.com/ros/robot_state_publisher/files/903734/Error.txt)

khalid2696, 2017-04-17:
After going through source code and other blogs on similar issues, I have found the issue with urdfdom_headers. There is an issue with the migration of shared pointers from boost to c++11. Downgrading the urdfdom_headers to version 0.4.2 did the trick. Similarly, urdfdom had to be downgraded to 0.4.2  
[https://github.com/ros/urdfdom_headers/issues/32](url)

sloretz, 2017-04-17:
Thanks for  posting what fixed it. I'll close this in favor of #59.

closed 2017-04-17 by sloretz

wallEVA96, 2017-04-17:
@ @sloretz   thank u, the day before yesterday  I have solve this with version 0.4.0,  I really compile, but i CANT use the rviz

wallEVA96, 2017-04-17:
thank u, the day before yesterday I have solve this with version 0.4.0, I really compile, but i CANT use the rviz @khalid2696