ros/robot_state_publisher #74: Lunar release

mikaelarguedas, 2017-04-11:
Hi @clalancette and @sloretz,

All dependencies of `robot_state_publisher` are now released in ROS Lunar :tada: !

Is it possible make a lunar release of `robot_state_publisher` soon to unlock [downstream packages](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22robot_state_publisher%22&s=6&r=1)? Thanks!

clalancette, 2017-04-11:
It looks like kdl_parser from robot_model is still failing on debian stretch.  Once that gets fixed, I'll do a release of robot_state_publisher into Lunar.

mikaelarguedas, 2017-04-11:
As mentioned on the [robot_model PR](https://github.com/ros/rosdistro/pull/14504#issuecomment-293164407), all robot_model Debian Stretch jobs fail because there is no Debian stretch branches on the robot_model-release repository. It needs to be rebloomed otherwise it'll keep failing.
I don't think it's preventing the release of robot_state_publisher in any way though. All the keys are satisfied and all the ubuntu jobs would pass on the first run.
@sloretz is on it

sloretz, 2017-04-11:
In progress rosdistro/pull#14508

mikaelarguedas, 2017-04-11:
awesome, thanks for the quick iteration!

sloretz, 2017-04-11:
Looks happy http://repositories.ros.org/status_page/ros_lunar_default.html?q=robot_state_publisher

closed 2017-04-11 by sloretz