ros/robot_state_publisher #76: Add C++11 flag to CMakeLists.txt

AuthenticEshkinKot, 2017-05-08:
robot_state_publisher depends from urdf, which uses std::shared_ptr and other abilities of C++11 in its headers. For example, shared pointer is used in /usr/include/urdf_world/types.h, so lack of appropriate compiler flag leads to errors.

sloretz, 2017-05-08:
Would you mind posting an example of an error you see?

AuthenticEshkinKot, 2017-05-09:
/usr/include/urdf_model/types.h:57:1: error: «shared_ptr» in namespace «std» does not name a template type URDF_TYPEDEF_CLASS_POINTER(Inertial);

cambesol, 2017-07-04:
I fixed this by adding the following in: 
src/ros_catkin_ws/src/robot_state_publisher/CMakeLists.txt
```
include(CheckCXXCompilerFlag)
check_cxx_compiler_flag(-std=c++11 HAS_STD_CPP11_FLAG)
if(HAS_STD_CPP11_FLAG)
  add_compile_options(-std=c++11)
endif()
```
This may not be the 'correct' solution, but I followed the same pattern as found in the urdf CMakeList.txt. I rased an issue on [https://github.com/ros/urdfdom_headers], but the problem lies with the robot_state_publisher build. All works now and I have successfully built Kinect on a Raspberry Pi 1 Model B with 120 packages included.

sloretz, 2017-07-05:
Does this only occur on raspbian?

sloretz, 2017-07-05:
The requirement for c++ 11 comes from urdfdom_headers after it switched from boost to std smart pointers. Since urdf urdfdom_headers is a headers-only library, there isn't a cmake target to get the compile options from, so @cambesol I think adding the requirement for C++11 to this library is the right fix.

[ROS Kinetic requires c++11](http://www.ros.org/reps/rep-0003.html), so the kinetic-devel branch could set the flag without checking.

cambesol, 2017-07-05:
I have only built this on Raspbian, so I cannot answer that question (assuming it was aimed at me). I like the check capability then add the flag rather than always setting it, but for Kinetic it makes sense with the changes to the urdfdom_headers. Sadly, for a single character change in a single file, it takes me ~5hours to see if it was successful fix.

sloretz, 2017-07-05:
@AuthenticEshkinKot If you get a chance, would you mind checking if #78 solves the issue?

clalancette, 2017-10-10:
We believe this is fixed now, so I'm going to close this out.  Feel free to reopen if the latest code still doesn't work for you.

closed 2017-10-10 by clalancette