ros/robot_state_publisher #79: Remove dependency on tf

sloretz, 2017-08-31:
It looks like robot_state_publisher already uses tf2 for publishing transforms, but depends on `tf` so it can use `tf::resolve` on a frame given a tf_prefix. Removing the tf dependency would required dropping support for tf_prefix ([which was removed in tf2](http://wiki.ros.org/tf2/Migration#tf_and_tf2)), and [would break ABI compatibility](https://github.com/ros/robot_state_publisher/blob/530bba1d7001afb2d61dea6c52b966111f927f20/include/robot_state_publisher/robot_state_publisher.h#L79) so it should be done no sooner than ROS Melodic.

andrestoga, 2018-02-13:
Hi,

I have a question: If robot_state_publisher uses tf2 for publishing transforms why is the topic message type of the topic /tf still tf/tfMessage instead of tf2_msgs/TFMessage?

best,

sloretz, 2018-02-14:
Hi @andrestoga, You'll probably get a better answer on https://answers.ros.org or on https://github.com/ros/geometry2 .

It may have something to do with the two message types being identical:
* [tf2 message](https://github.com/ros/geometry2/blob/indigo-devel/tf2_msgs/msg/TFMessage.msg)
* [tf1 message](https://github.com/ros/geometry/blob/indigo-devel/tf/msg/tfMessage.msg)

sloretz, 2018-02-14:
Fixed by #82

closed 2018-02-14 by sloretz