ros/robot_state_publisher #81: robot-state-publisher crashes on an empty URDF

sloretz, 2018-02-13:
Create a launch file with the following content. It initializes `robot_description` to contain a urdf that has no links.

```
<?xml version="1.0"?>
<launch>
  <param name="robot_description" value='&lt;robot name="foo" xmlns:xacro="http://ros.org/wiki/xacro"&gt;&lt;/robot&gt;'/>
  <node pkg="robot_state_publisher" name="robot_state_publisher" type="robot_state_publisher" output="screen"/>
</launch>
```

`robot_state_publsiher` crashes with the following output.

```
process[robot_state_publisher-2]: started with pid [316]
[ERROR] [1518481302.405483470]: No link elements found in urdf file
[robot_state_publisher-2] process has died [pid 316, exit code -11, cmd /opt/ros/indigo/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/developer/.ros/log/dd4738a0-1053-11e8-81e1-0242ac110005/robot_state_publisher-2.log].
log file: /home/developer/.ros/log/dd4738a0-1053-11e8-81e1-0242ac110005/robot_state_publisher-2*.log
```

The node should exit in this case since there's nothing it can do with an empty URDF, but it shouldn't crash.

betab0t, 2019-01-15:
Easy fix :smile:. @sloretz i hope #97 solves the issue.
seems like the node exit with generic error code/status -1 if it failes to load URDF, regardless of the specific resone, so i added a check for no links case, now it will exit cleanly(status 0). 
Also, since [error message anyways printed by urdfdom lib](https://github.com/ros/urdfdom/blob/master/urdf_parser/src/model.cpp#L179) i didnt feel the need to add extra error/warning about it...

sloretz, 2019-01-17:
Ooops, Looks like this is only an issue on Indigo https://github.com/ros/robot_state_publisher/pull/97#discussion_r248428697

Since indigo EoL is approaching, there likely won't be another release into it. I'll go ahead and close this issue. Sorry for the confusion.

closed 2019-01-17 by sloretz