ros/robot_state_publisher #90: use_sim_time = false makes /robot_state_publisher die!

rahulbhadani, 2018-07-24:
In ROS Melodic on Ubuntu 18.04 LTS, when I use  `<arg name="use_sim_time" default="false"/>`, then robot state publisher dies with following messages:

t```
erminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
[catvehicle/robot_state_publishercatvehicle-3] process has died [pid 5998, exit code -6, cmd /opt/ros/melodic/lib/robot_state_publisher/robot_state_publisher /joint_states:=/catvehicle/joint_states __name:=robot_state_publishercatvehicle __log:=/home/ivory/.ros/log/8f8c35ca-8f15-11e8-b7c7-8c1645b122d9/catvehicle-robot_state_publishercatvehicle-3.log].
log file: /home/ivory/.ros/log/8f8c35ca-8f15-11e8-b7c7-8c1645b122d9/catvehicle-robot_state_publishercatvehicle-3*.log

```
You can refer to https://github.com/rahulbhadani/catvehicle for the package I am using. 

In catvehicle_empty.launch, set  `<arg name="use_sim_time" default="false"/>`
Step 1. `roslaunch catvehicle catvehicle_empty.launch` 
Step 2. `roslaunch catvehicle catvehicle_custom.launch`

sloretz, 2018-07-24:
Hi @rahulbhadani, would you mind providing a [minimal working example](https://en.wikipedia.org/wiki/Minimal_Working_Example) example?

I'm unable to reproduce with

```xml
<?xml version="1.0"?>
<launch>
  <param name="robot_description" textfile="$(find robot_state_publisher)/test/two_links_moving_joint.urdf" />

  <node pkg="robot_state_publisher" name="two_links_moving_joints_pub" type="state_publisher">
    <param name="use_sim_time" value="false"/>
  </node>
</launch>
```

rahulbhadani, 2018-08-05:
I gave you the link og github repo. You can follow that to reproduce the scenario.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)