ros/robot_state_publisher #91: sdf support and example for robot_state_publisher

rahuldeo2047, 2018-12-26:
I need to work on sdf due to cyclic parent-child relation of links not supported in urdf. I could not find and example with sdf.

Let me know if you need any more detail.

sloretz, 2019-01-22:
Hi @rahuldeo2047, correct me if I've misunderstood, it sounds like this is a request for `robot_state_publisher` to publish `tf` transforms given an `sdf` description of a robot. Given that `tf` transforms must form a tree, what transforms should be published for a robot with a cycle?

sloretz, 2020-10-12:
I'll close this due to inactivity. There has been progress on using SDFormat with `robot_state_publisher` in the pull request here: https://github.com/ros/sdformat_urdf

closed 2020-10-12 by sloretz

rahuldeo2047, 2020-10-12:
If I remember correctly. Yes. You are correct.
I will get back with my memory refreshed.

The package is well supported for urdf based tf. And able to visualise the tf along with robot in rviz. However, not with sdf based robot.
I will get back with more context it is not the suitable one. 

Thank you.