ros/robot_state_publisher #99: BSD or LGPL?

andrewbacha, 2019-02-06:
The robot_state_publisher package is marked as BSD, but treefksolverposfull_recursive.cpp and treefksolverposfull_recursive.hpp have header comments identifying those files as LGPL. Since this source code gets compiled in with the rest of the code and cannot be separated via dynamic linking, I don't see how the whole package can be used without complying with the LGPL license.

clalancette, 2019-02-06:
Good point, thanks for bringing it up.  We're looking into the provenance of the code, we'll update this issue (and/or the license field) once we figure out exactly what the license should be here.

andrewbacha, 2019-02-06:
Great! Thanks for the quick response and looking into clearing up the confusion.

clalancette, 2019-02-08:
It turned out that the LGPL code was never actually used in this package.  Thus, we've removed that code in #100, #101, and #102 so that this ends up being a pure BSD package.  With that, I'll close this out, but feel free to keep commenting if you need more clarification.

closed 2019-02-08 by clalancette