ros/robot_state_publisher #108: Release from melodic-devel branch

sloretz, 2019-03-19:
I was wondering why no PR job ran for #103, and it looks like it was released to melodic from the `kinetic-devel` branch

https://github.com/ros/rosdistro/blob/25b9046eda193ed47cbd325b1a046f40ffba75de/melodic/distribution.yaml#L4931-L4946

I haven't looked closely at whether melodic-devel has breaking changes, but a re-release on the melodic branch should be considered.

clalancette, 2019-03-20:
You are totally right; not sure how I messed that one up.  I think we'll need to actually look at cross-porting changes between kinetic-devel and melodic-devel to make sure we have everything.

clalancette, 2019-03-22:
So doing some research, it looks like https://github.com/ros/robot_state_publisher/pull/82 was the reason we wanted to release from a different branch.  Otherwise, the kinetic-devel branch and melodic-devel branch are broadly similar.

I'll point out that when we merged #82, there were only 2 PRs necessary to fix downstream breakage.  One of them was just lack of a header file, which was merged and which really wasn't directly involved.  The other was https://github.com/leggedrobotics/xpp/pull/5, which was closed because we never did break the API and thus it didn't apply.

So I see 3 different ways we can handle this from here:
1.  We can either knowingly break the API of robot_state_publisher by release from melodic-devel as-is with #82 applied.  There aren't a lot of downstream consumers that subclass it, so the breakage would probably be minimal (but not zero).
1.  Revert #82 completely from the melodic-devel branch, which will restore the API there and make it so that things worked as they once did.
1.  Revert the API change from #82, but don't do the tf_prefix resolving internally anymore.  We'll have a vestigial API parameter, but we can remove that in Noetic.

@sloretz Thoughts?

jschleicher, 2019-05-15:
I'd strongly encourage to get rid of the tf dependency in melodic (as it clutters minimal images with e.g. graphviz https://github.com/ros/geometry/issues/189).

So I prefer option 1.

jschleicher, 2019-07-12:
@sloretz @clalancette Any progress or decision?