ros/robot_state_publisher #110: Could not load the file sensor_msgs/JointState.h

r-hede, 2019-05-20:
Hi,

I have an error with the package `robot_state_publisher` when I'm try to `catkin_make_isolated`.

Specifically, I get an error from the file `robot_state_publisher/include/robot_state_publisher/joint_state_listener.h` on line `43`.

`fatal error: sensor_msgs/JointState.h: No such file or directory`

However, the dependency `sensors_msgs` is correctly loaded in the `package.xml` file and the searched file exists at the indicated path.

Could you help me ?

Best regards,

Robin Hède.

r-hede, 2019-05-24:
I finally solved my problem by using another package path.

Instead of using the "ros_comm" package path, I used "robot".

`rosinstall_generator robot --rosdistro melodic --deps --tar > melodic-robot.rosinstall`

The [robot variant](http://www.ros.org/reps/rep-0131.html#robot-variant) is defined to be core, stable, ROS libraries for any robot hardware. It is the "general robotics" libraries of ROS. It may not contain any GUI dependencies.

The complete procedure to installing the Melodic version of ROS from sources is available [here](https://wiki.ros.org/melodic/Installation/Source).

closed 2019-05-24 by r-hede

gavanderhoorn, 2019-05-24:
> This might fit better as a question on https://answers.ros.org .

This was actually already cross-posted to ROS Answers: [Could not load the file sensor_msgs/JointState.h](https://answers.ros.org/question/323534).