ros/robot_state_publisher #119: ROS2: Switch back to parameter for description?

clalancette, 2019-10-08:
Original issue is at https://github.com/ros2/robot_state_publisher/issues/33.  Here's the thread:

---
jacobperron:

The ROS 1 version uses a parameter to read the robot description. The ROS 2 version reads the description from a file, with the path provided as a command line argument. Two questions:

1. What was the reason for this change? I guess it was related to parameters not being available in ROS 2 at the time.

2. Are there plans to switch back to using a parameter?

I ask because this has impact for users porting launch scripts from ROS 1. For example, (I think) typical usage with xacro in a launch file would look something like this:

    <launch>
      <param name="robot_description" command="$(find xacro)/xacro '$(find foo_description)/urdf/foo.urdf.xacro'" />
      <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    </launch>

Translating this file into a ROS 2 launch script, I would expect it to look like:

    <launch>
      <node name="robot_state_publisher" pkg="robot_state_publisher" exec="robot_state_publisher">
        <param name="robot_description" cmd="$(exec-in-pkg xacro xacro) '$(find-pkg-share foo_description)/urdf/foo.urdf.xacro'" />
      </node>
    </launch>

Ignoring the fact that `xacro` isn't ported to ROS 2 yet and the `cmd` attribute isn't a valid attribute for the `<param>` tag (but it should be I think), I expect many people will run into the challenge of updating launch files similar to this example.

Is there an alternative workflow you imagine?

---
clalancette:

>     1. What was the reason for this change? I guess it was related to parameters not being available in ROS 2 at the time.

@Karsten1987 Did the initial port here.  I'm guessing that the current state is because there were no parameters at the time, but I'm not sure.

> 
>     2. Are there plans to switch back to using a parameter?

Personally, I'm on the fence with it.  I sort of think of parameters as "optional" things you might configure on a node.  In this case, the description is really core to what `robot_state_publisher` does, so it kind of makes sense that you have to provide it as a command-line parameter at startup time.  There is another argument here that you may want to launch `robot_state_publisher` without any description and then provide one later (this is sort of the base case of having the description be dynamic at runtime).  But I would argue that you still don't want to use a parameter, since it is somewhat hard to notify all downstream users of changes.  You can do it via the `/parameter_events` topic, but then you have to filter out all of the non-`robot-description` updates in your downstream subscriber callbacks, or use a namespace.  It seems to be simpler to just have a topic and a service on `robot_state_publisher`; `current_description` (latched), and `set_description`.  That way it is easy to update all of the downstream consumers.

All of that being said, these are just some thoughts floating around my head, and I haven't really run them by anyone yet.  So there may be significant downsides to doing something like that.  Let me ask a different question: other than the porting effort, do you see a significant problem with the current state of affairs?

---
sloretz:

> 2. Are there plans to switch back to using a parameter?

A parameter is only on one node, but multiple nodes may need it (ex: [MoveIt `RobotModel()`](http://docs.ros.org/kinetic/api/moveit_core/html/classmoveit_1_1core_1_1RobotModel.html#ab2d8a71b22b1659634434f9badaa706b) needs a URDF). I think this is why `robot_state_publisher` [publishes a "latched" `robot_description` topic](https://github.com/ros2/robot_state_publisher/blob/dac0072ea1128863a6327c10902cf57036657ce8/src/robot_state_publisher.cpp#L86). I don't think there are plans to change, but a good option may be to make `robot_state_publisher` a consumer of the `robot_description` topic published by a brand new `robot_description_publisher`. That would enable someone to load descriptions from someplace other than the filesystem by replacing `robot_description_publisher` without needing to reimplement `robot_state_publisher` too. It would also make it easier to enable `robot_state_publisher` to handle changing robot descriptions, since it would have to do something when it received a message.

---
jacobperron:

Don't get me wrong, I think current latched publisher makes sense. What I think is awkward with the current implementation is that we have to start `robot_state_publisher` by [loading a URDF from a file](https://github.com/ros2/robot_state_publisher/blob/dac0072ea1128863a6327c10902cf57036657ce8/src/joint_state_listener.cpp#L223). It would be better, IMO, to read a URDF as a string (like in ROS 1), or support both options.

The issue with the current state of affairs is that I can't run a tool like xacro (which can output URDF as a string) and pass it to robot_state_publisher. E.g.

    ros2 run robot_state_publisher robot_state_publisher $(xacro /path/to/my/model.urdf)

It requires a two step process, first write the output of xacro to a file and then pass it to robot_state_publisher. I don't have a strong opinion on if we use a parameter (like in ROS 1) or a regular CLI arg to pass the URDF as a string, but it seems like a valid use-case to me.

---
clalancette:

I see the problem, and yeah, that sucks.

But I don't think a parameter is the solution.  I discussed this with @IanTheEngineer, and we came up with a different solution.  What if we instead allowed `-` as the filename, and then took the URDF from stdin?  That would be more Linux-y, solve this particular problem, and be relatively easy to implement (there is a slight concern about this working on Windows, but I think we can figure something out).  We'd also probably want to add some launch support syntax for it, though that can come later.  Thoughts?

---
[... 73 lines cut ...]
clalancette:

> I'm one of the users Jacob was originally talking about. The trouble I see with the current state of affairs: writing xacro results to a file and then loading them is awkward and extra effort,

Agreed.  My proposal of using the stdin trick for the file would allow you to pipe the results from xacro directly into robot_state_publisher.

> robot_state_publisher publishes an empty /robot_description topic

That shouldn't be.  It will currently only publish once it has gotten its first update; are you sure that is happening?

> and I can't figure out any syntax for feeding xacro results to robot_state_publisher in an XML launch file.

I think we'd be able to do something with the XML launch file and the stdin trick, but I'll have to check on that.

> 
> Having dynamic robot descriptions sounds great, but it also sounds like a huge can of worms. I currently only need static robot descriptions and would love to have easy solutions for spawning robots in both python and XML right away.

Sure, but it is a huge can of worms that we have been kicking down the road for years.  I'm not saying that I want to solve the dynamic problem today, but I do want to make it so that the solution we start here at least allows a little more dynamicity.  I also want to make sure that we are using features as they are semantically intended, and it is not clear to me that the description as a parameter is that.

---
mogumbo:

I don't know what kind of update robot_state_publisher is expecting. I see that it subscribes to /joint_states, but no information will be generated on that topic until after the robot is spawned and libgazebo_ros_joint_state_publisher.so starts publishing.

Anyway, the particulars of this problem are a bit out of scope for this thread. Maybe I should take it elsewhere.

jdlangs, 2019-10-21:
I've been thinking about this problem for some time and would have chimed in a while ago if I had caught this thread earlier. I think this is a great opportunity to make some fundamental improvements instead of just trying to get back to exactly how we did things in ROS1. IMO, the core issue is `xacro`, specifically that it operates over stdin/stdout. At some point, a file needs to be read off the disk and I think it makes more sense for it to be read by the node that's going to provide the information to the rest of the system, rather than as a side effect of the launching process. This could be either a `robot_state_publisher` or a `robot_description_publisher` node that then calls xacro programmatically. I especially like the standalone description publisher node idea because it can be indefinitely extended as the authoritative source of information about an individual robot in a ROS system. Eventually it could probably be more-or-less fully merged with the contents of MoveIt config packages.

Are there other use-cases besides xacro that wouldn't work with a node reading a robot description file from the disk?

clalancette, 2019-10-21:
I've been (slowly) doing a full revamp of the code in robot_state_publisher for ROS 2; you can see the current state of my work at https://github.com/ros/robot_state_publisher/tree/ros2-refactor .  At the moment, I feel like the code is largely done; what's left is tests and documentation (though we need a *lot* of both, so it may take some time to finish).

One thing that came out of doing the implementation is that we actually do need to pass the initial description as a parameter.  The main reason for that is that we really want to make the `RobotStatePublisher` class composable, which means that the constructor can't take any arguments besides an `rclcpp::NodeOptions` structure.  So on that branch I have indeed made `robot_description` a parameter, as is done in ROS 1.

@jdlangs In theory, we could add another parameter like `robot_description_xacro_file` to read in the file, execute xacro on it, and then just use it.  The problem with that is that xacro is completely written in Python, while robot_state_publisher is completely written in C++.  Thus, the only way that I can see to call into xacro is to either embed a Python interpreter in robot_state_publisher (yuck), or call xacro via `system` (double-yuck).  Unless you have another idea, I don't think we can do this.

One possible thing to do would be to add a parameter called `robot_description_file`, which is just expected to be a valid XML file.  robot_state_publisher could read either `robot_description` or `robot_description_file` at startup, and if it is the latter, read the contents and use it.  For the xacro use-case, then, you could arrange for the launch file to call xacro and dump the file to disk, then set the parameter for `robot_description_file` to point to it, achieving your goal in a slightly different way.

Thoughts?

sloretz, 2019-10-21:
> One thing that came out of doing the implementation is that we actually do need to pass the initial description as a parameter.

It seems like a ROS 2 parameter may not be the best fit since they are specific to a node. In ROS 1 it's the opposite, the `/robot_description` parameter is available to all nodes via the global parameter server. If the standard way to pass `robot_description` in ROS 2 is a parameter, then it needs to be repeated for every other node. Maybe every node has its own `robot_description` parameter, and the expectation is they're all launched with the same copy? Maybe every node is given the name of another node they need to query for the value of it's `robot_description` parameter?

>  The main reason for that is that we really want to make the RobotStatePublisher class composable, which means that the constructor can't take any arguments besides an rclcpp::NodeOptions structure.

I think `robot_state_publisher` subscribing to a `topic` that it receives the description on would be composible too. A topic seems like a better way for multiple nodes to get the same info. The publisher of that topic could still read the description from a parameter, or from disk, or from stdin, etc.

clalancette, 2019-10-21:
> I think `robot_state_publisher` subscribing to a `topic` that it receives the description on would be composible too. A topic seems like a better way for multiple nodes to get the same info. The [...]

You do bring up a good point here about multiple nodes needing the description.  I've been thinking about this as well, and in a ROS 2 world, I actually think it makes sense for robot_state_publisher to be the "owner" of the robot_description.  That is, any time a node wants the robot_description, they have to get it from robot_state_publisher.

In my current work on the `ros2-refactor` branch, the robot_description is initially provided to robot_state_publisher as a parameter.  An outside node *could* get that description via a service call to get parameters from robot_state_publisher.  But even better is that robot_state_publisher also has a transient local topic called `robot_description`, where it outputs the description.  Thus outside nodes can just subscribe to the `robot_description` topic, and get the information they need along with any updates.

So in this proposed world, the description is initially provided to robot_state_publisher as a parameter, but is disseminated to the rest of the graph via a transient_local topic.  Thoughts?

clalancette, 2019-11-15:
To revive this issue, I have a couple of updates.

1.  The current state of the [ros2-refactor](https://github.com/ros/robot_state_publisher/tree/ros2-refactor) branch does switch back to using a parameter for the `robot_description`.  This is necessitated because of the composed node case, basically.

1.  I've posted a message to discourse discussing the current state of my rewrite: https://discourse.ros.org/t/ros-2-robot-state-publisher-improvements/11521 .  Only part of it applies to this issue (which is how we pass the initial description), but anyone is welcome to comment there.

1.  Separately, I've been doing some testing with my rewrite to see how we could generate the URDF with something like `xacro` and pass it to `robot_state_publisher` without a temporary file.
    * If using Python launch files, this is actually very straightforward; you `import xacro`, call the appropriate APIs to convert the xacro file to a URDF string, and then pass that along in the parameters.
    * If using the XML launch frontend, it is actually not possible to do this in a generic way at present.  There is no XML launch mechanism to call out to an arbitrary program, capture the output, and stuff it into a variable.  Thus, for this to work we'll need an update to the XML launch frontend as well.

jacobperron, 2020-01-08:
Related issue https://github.com/ros2/launch/issues/366

clalancette, 2020-04-02:
We ended up indeed switching to a parameter here in #126 , so closing this out.

closed 2020-04-02 by clalancette