ros/robot_state_publisher #120: ROS 2: switch the license back to BSD

clalancette, 2019-10-11:
The original ROS 2 port of this package switched the license to Apache 2.0, but it is clearly derived from the original ROS port which is under the BSD license.  Thus, the license on the ROS 2 port should be switched back to BSD.  Looking at the contributor list, it is largely OSRF people who contributed, but there are a few external contributors:

@pohzhiee
@pbeeson
@crdelsey
@bponsler

Could you reply here and tell us if you are OK with us relicensing your original contributions back to BSD?  Thanks in advance.

bponsler, 2019-10-11:
I have no objections

pohzhiee, 2019-10-11:
No objections

crdelsey, 2019-10-11:
I have no objections either.

pbeeson, 2019-10-12:
No objections from me

clalancette, 2019-10-14:
Thanks so much for prompt responses!  I'll open a PR changing this back to BSD shortly.

clalancette, 2019-10-14:
Hm, not sure why github didn't auto-close this, but now that I've merged #121, this can be closed.  Thanks again everyone!

PR #121 (2019-10-14): Switch the license back to BSD.
The 'ros2' branch of robot_state_publisher is clearly derived
from the ROS 1 port (that lives on the 'indigo-devel' and
'melodic-devel' branches).  Thus, the license should be BSD.
There were a few community contributions to the 'ros2' branch
under the (erroneous) Apache 2.0 license, but all contributors
have signified their willingness to license back to BSD in
https://github.com/ros/robot_state_publisher/issues/120 .
This change switches the license back to the correct BSD
license on the 'ros2' branch.

@Karsten1987 FYI.

Fixes #120
closed 2019-10-14 by clalancette