ros/robot_state_publisher #131: Publishing tf at with future timestamp if it's not static

weihunko, 2020-02-17:
in `void RobotStatePublisher::publishFixedTransforms()`,  if the frame is not static, it will publish the tf with the timestamp that is 0.5 second in the future.
--> https://github.com/ros/robot_state_publisher/blob/kinetic-devel/src/robot_state_publisher.cpp#L117

Can anyone elaborate the design decision behind this?

wjwwood, 2020-02-19:
This sounds familiar, but I cannot remember the reason, @tfoote, @clalancette, or @sloretz, do you remember why this happens?

wjwwood, 2020-02-19:
Maybe this? https://answers.ros.org/question/218864/why-does-amcl-post-date-tf-transform_tolerance/

wjwwood, 2020-02-19:
Ah, it's mentioned here on the wiki:

https://wiki.ros.org/robot_state_publisher

> Note: All fixed transforms are future dated by 0.5s. 

which links to https://wiki.ros.org/tf/FAQ#Why_do_I_see_negative_average_delay_in_tf_monitor.3F

closed 2020-02-19 by wjwwood