ros/robot_state_publisher #142: ROS2 Dashing tf ignores namespace

joe28965, 2020-06-11:
Normally in ROS Dashing you can parse the argument `__ns:=/namespace` to add namespaces to the topics when they're created. If you try this for the robot_state_publisher, the topic of tf remains /tf instead of /namespace/tf. The name of the node and the other topics do use the namespace given to them, but tf ignores this. I was wondering if others have noticed this same issue, and why this is.

bastinat0r, 2020-07-08:
I currently have the same problem. I want to run multiple robots (in namespaces, or using tf-prefix which is deprecated according to #125 ).  I really do not know how I should design the system using the new ros2 ecosystem.

joe28965, 2020-07-08:
I ended up switching to eloquent myself

bastinat0r, 2020-07-08:
I use foxy, now I am really confused ;) But I just remapped the /tf and the /tf_static topic to the namespace and it is working (for now).

clalancette, 2020-07-08:
> I use foxy, now I am really confused ;) But I just remapped the /tf and the /tf_static topic to the namespace and it is working (for now).

The way that this works is that `/tf` and `/tf_static` are global topics, meaning that they ignore namespaces.  However, you can *always* remap topics, even global ones.  So it is perfect fine to remap them by hand, and that will continue to work in the future.

sloretz, 2020-10-12:
`/tf` and `/tf_static` are global topics, meaning they don't get namespaced. To remap them, use the rules `/tf:=/namespace/tf` and `/tf_static:=/namespace/tf_static`. See the [ROS 2 remapping design doc](http://design.ros2.org/articles/static_remapping.html) for more info. I'll close this since it doesn't appear to be an issue with `robot_state_publisher`.

If you're looking for an alternative to the `tf_prefix` behavior, please see #125 for a ROS 2 ticket.

closed 2020-10-12 by sloretz