ros/robot_state_publisher #147: ROS Noetic multi-robot system launch file

PedroACaldeira, 2020-09-10:
I've been trying to create a launch file for a multi robot system in noetic but I have issues with the robot_state_publisher package because I can't use the tf_prefix and as such, I have no way of organizing my robot system. I would like to know if there are any work around to make the launch file operable? For reference, I also made a question in the ROS Answers forum: https://answers.ros.org/question/360480/multi-robot-simulation-on-gazebo/

RaphvK, 2020-09-25:
See #139

oKermorgant, 2020-09-29:
Hi, 
I am facing the same issue while porting some tutorials / labs on ROS 2.

I hope the `tf_prefix` support pops up at some time. Meanwhile, a workaround that brings similar features is to use xacro prefixes for links. Basically all references to links in the URDF file are replaced by `"$(arg prefix)base_link"`.
The xacro file also takes an argument `<xacro:arg name="prefix" default=""/>`. 
You get the corresponding URDF by calling `xacro <file> prefix:=<prefix>/`.

If you are not the owner of the URDF / xacro files that you use, unfortunately it makes it mandatory to use a modified version. It would indeed be much simpler to have `robot_state_publisher` able to add a prefix on-the-fly without having to use a custom xacro file.

sloretz, 2020-10-12:
FYI some comments on bringing prefixing behavior back as a `robot_state_publisher` feature here: https://github.com/ros/robot_state_publisher/pull/139#pullrequestreview-506804970

SystemDiagnosticss, 2021-04-28:
Any updates in this case?

sloretz, 2021-09-30:
Resolved in #169

closed 2021-09-30 by sloretz