ros/robot_state_publisher #148: Command substitution failing for ros2 branch: rsp-launch-xacro-command-subst.xml

sjauhri, 2020-09-23:
When trying to launch using the xml file 'rsp-launch-xacro-command-subst.xml' on the ros2 branch, I get the following error:

**Error trying to process rule "substitution":
Unknown substitution: command**

Relevant launch file code:
<node pkg="robot_state_publisher" exec="robot_state_publisher">
    <param name='robot_description' value="$(command 'xacro $(find-pkg-share robot_state_publisher)/urdf/test-desc.urdf.xacro')"/>

I'm running ros2-eloquent.
This is strange because eloquent should already have the command substitution feature https://github.com/ros2/launch/issues/366

clalancette, 2020-09-23:
> This is strange because eloquent should already have the command substitution feature [ros2/launch#366](https://github.com/ros2/launch/issues/366)

No, that was merged in for Foxy, not Eloquent.  With Eloquent going out-of-support in 2 months, we're unlikely to backport it.

I'm going to go ahead and close this out, but feel free to keep commenting or reopen.

closed 2020-09-23 by clalancette

sjauhri, 2020-09-23:
Ooh allright then.
What's the alternative for doing this in Eloquent then?

clalancette, 2020-09-23:
It's been a while since I tried this in Eloquent.  I think you have to arrange to call the xacro generation yourself, then pass the full-formed URDF.  I don't think you can do that through the XML launch files in Eloquent; you'd have to switch to Python launch files.

Other ideas for Eloquent:
1.  Generate the URDF by hand (call `xacro ...` on the command-line), then check that into your git repository.  During install time, install that, and then point your launch files at that.
1.  Generate the URDF at CMake build time, and install that.  Then point your launch files at that.

There may be others.  If none of the above solutions work for you, I'll suggest opening a question at https://answers.ros.org since someone else might have additional ideas.