ros/robot_state_publisher #152: Cannot use tf_prefix parameter when update from ROS Melodic to ROS Noetic

ngkhiem97, 2020-10-11:
[image]

In tf view_frames robot_state_publisher does not publish the tf_prefix of the robots even though I included in the launch file.

`<node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher" output="screen">
      <param name="tf_prefix" value="$(arg tb1)" />
      <param name="publish_frequency" type="double" value="50.0" />
    </node>`

droemer7, 2020-10-12:
See #139. Noetic dropped tf_prefix, and #139 is requesting it to be restored. You may want to comment over there.

sloretz, 2020-10-12:
Thanks for the issue! I'll close this as a duplicate of #147. There's a similar issue for ROS 2 in #125

closed 2020-10-12 by sloretz