ros/robot_state_publisher #153: Ros2 foxy symbol lookup error issue

Fricodelco, 2021-01-17:
I use ros2 foxy with arm64 ubuntu and tried to use robot_state_publisher. Get this error:
/home/rodion/tb3_ros2/install/robot_state_publisher/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /home/rodion/tb3_ros2/install/robot_state_publisher/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev

sloretz, 2021-01-19:
Is this while building from source? If so, what command are you using to build `robot_state_publisher`?

SEUZTh, 2021-09-01:
> Is this while building from source? If so, what command are you using to build `robot_state_publisher`?

Hello, I met the same problem in my parallels ubuntu 20.04 arm.
I installed ROS 2 via Debian Packages.
`roscore` is ok, but when I used `ros2 run demo_nodes_cpp talker`, I received:
```
/opt/ros/foxy/lib/demo_nodes_cpp/listener: symbol lookup error: /opt/ros/foxy/lib/demo_nodes_cpp/listener: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
```
However, `ros2 run demo_nodes_cpp add_two_ints_client` works well.

clalancette, 2021-09-01:
So the fact that you are running `roscore` makes me think you are mixing ROS 1 and ROS 2.  In that case, the library loading might be getting confused between ROS 1 class loader and ROS 2 class loader.  If you only source ROS 2 foxy, does the problem go away?

SEUZTh, 2021-09-01:
> So the fact that you are running `roscore` makes me think you are mixing ROS 1 and ROS 2. In that case, the library loading might be getting confused between ROS 1 class loader and ROS 2 class loader. If you only source ROS 2 foxy, does the problem go away?

Thanks for your reply, I tried to select ROS1 or ROS2 when I open a new terminal through `.bashrc`. 
```
# select ros1 or ros2 in terminal
echo "ros noetic(1) or ros2 foxy(2)?"
read edition
if [ "$edition" -eq "1" ];then
  source /opt/ros/noetic/setup.bash
else
  source /opt/ros/foxy/setup.bash
fi
```
But it seems that the version of ROS haven't been changed. After I wrote `source /opt/ros/foxy/setup.bash` in `.bashrc` , the problem was solved.

clalancette, 2021-11-12:
Closing based on the last comment.

closed 2021-11-12 by clalancette

hussains72, 2024-04-04:
# select ros1 or ros2 in terminal
echo "ros noetic(1) or ros2 foxy(2)?"
read edition
if [ "$edition" -eq "1" ];then
  source /opt/ros/noetic/setup.bash
else
  source /opt/ros/foxy/setup.bash
fi

to avoid such kind of error you must put above script at the end of .bashrc file

hussains72, 2024-04-04:
`# select ros1 or ros2 in terminal
echo "ros noetic(1) or ros2 foxy(2)?"
read edition
if [ "$edition" -eq "1" ];then
  source /opt/ros/noetic/setup.bash
else
  source /opt/ros/foxy/setup.bash
fi`

to avoid such kind of error you must put above script at the end of .bashrc file

hussains72, 2024-04-04:
`# select ros1 or ros2 in terminal  
echo "ros noetic(1) or ros2 foxy(2)?"  
read edition  
if [ "$edition" -eq "1" ];then  
  source /opt/ros/noetic/setup.bash  
else  
  source /opt/ros/foxy/setup.bash  
fi`

hussains72, 2024-04-04:
`# select ros1 or ros2 in terminal  
echo "ros noetic(1) or ros2 foxy(2)?"  
read edition  
if [ "$edition" -eq "1" ];then  
  source /opt/ros/noetic/setup.bash  
else  
source /opt/ros/foxy/setup.bash
fi`  

to avoid the error you must put this script at the end of the .bashrc file