ros/robot_state_publisher #155: Parameter robot_description as file path, not URDF string

karl-schulz, 2021-02-06:
Hi!

When providing a URDF as a argument (now deprecated), we could provide a path to a URDF file. This has been replaced by a parameter, which makes sense. However, this parameter now has to store a full XML description, which is not possible to store in a YAML file. 

* How is this supposed to work? 
* Is there a reason that the node now doesn't accept file paths anymore? (I suppose so, but it's not clear to me)

Anybody I see using robot_state_publisher still uses the deprecated argument to provide the URDF.

Thanks!

clalancette, 2021-02-08:
> Is there a reason that the node now doesn't accept file paths anymore? (I suppose so, but it's not clear to me)

The main reason that it has to be a parameter and not a command-line option is for the case where you want to [compose](https://index.ros.org/doc/ros2/Tutorials/Composition/) robot_state_publisher with other modules.  In that case, there is no command-line at all.  Since you have to deal with that case anyway, you may as well just always use the parameter.

> How is this supposed to work, especially with dynamic URDF files, e.g. in launch files and using substitutions?

There are a bunch of examples on how to make this work in the repository here: https://github.com/ros/robot_state_publisher/tree/ros2/launch .

In the future, questions like this are better asked on https://answers.ros.org, so that they are in one searchable place.  I'm going to close this out (as there is no real change to be made with this), but feel free to reopen or keep commenting.

closed 2021-02-08 by clalancette

karl-schulz, 2021-02-08:
Hi @clalancette , thanks for your quick answers!

But I was not wondering why it doesn't accept command-line arguments, but why it doesn't accept file names. How it is now, the URDF  file has to be read before then providing its content as a parameter - this means the expression has to be evaluated, which is not possible in a launch file with dynamic substitutions. 

In the examples you gave, the param files are all hardcoded in the launch files. So it is not possible to use substitutions anymore, right? Am I missing something?

karl-schulz, 2021-02-08:
I will post this question on https://answers.ros.org/ as you said.

clalancette, 2021-02-08:
> But I was not wondering why it doesn't accept command-line arguments, but why it doesn't accept file names.

Oh, I see.  Well, that is a longer and more complicated answer.

One of the (eventual) goals is for the ROS 2 ecosystem to be able to support URDFs that change on the fly.  Think of a manipulator that can choose different end-effectors; when the manipulator chooses a different end-effector, you want to update the URDF.  In order to do that, you need to have one source of truth for the URDF in the system.

And while `robot_state_publisher` *could* be that source of truth, it isn't necessarily that source.  We've had people who want to provide that source of truth via a simulator, for instance, in which case `robot_state_publisher` would be a receiver of that source of truth, not the provider.  Does that make sense?

> this means the expression has to be evaluated, which is not possible in a launch file with dynamic substitutions.
> In the examples you gave, the param files are all hardcoded in the launch files. So it is not possible to use substitutions anymore, right? Am I missing something?

I guess I don't totally understand what you are talking about here.  Can you give a concrete example of what you are trying to accomplish?

karl-schulz, 2021-02-09:
Ok, I understand. Thanks!

Regarding the substituions, I meant the `Substitution` from the `launch` package for creating Python launch scripts. Let's say I have two slightly different robots (e.g. Robot versions 1.1 and 1.2) with slightly different dimensions, however sharing the other properties. I'd like to have similar launch files for both, but they should be able to dynamically load a different URDF file for versions 1.1 and 1.2, respectively.

Turtlebot does this too for its different versions, but solves it with a environment variable TURTLEBOT3_MODEL, which I think is not ideal:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py

Previously, one could create a launch file with a `LaunchConfiguration("urdf_file")` and `DeclareLaunchArgument("urdf_file")`, or any other kind of substitution. And then this launch file could either be included by other launch files providing a "urdf_file" argument, or the launch file could get its URDF file from the command line. This LaunchConfiguration then gets evaluated when launching the file with ros2 launch *.

But now, as the URDF file already has to be read when the `generate_launch_description()` is called, meaning before the `LaunchConfiguration` etc. are evaluated. That means a launch file which uses robot_state_publisher needs to be copy-pasted for every robot model, only with differing URDF file paths.

To me it seems that a lot of flexibility is lost, when we need to provide the file contents as the parameter. Would it be possible to allow this option, e.g. in a `robot_description_file` parameter? This parameter could then be generated by a substitution. I would also create a pull request, if you think that makes sense.

I hope I could explain what I meant, let me know if not.

karl-schulz, 2021-02-09:
Okay, so I was able to get the desired behavior working.

Old launch file:
```
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

def generate_launch_description():
    conf_model = LaunchConfiguration('model')
    return LaunchDescription([
        Node(
            package="robot_state_publisher",
            name="robot_state_publisher",
            executable='robot_state_publisher',
            arguments=[conf_model],   # Pass as argument
            output="screen",
        ),
        # ... launch other nodes, etc.
    ])
```

To load a file dynamically with a Substitution, I needed to wrap it in a Command. But then, the yaml contents were not excaped correctly (that is only enabled for 1st layer pure string parameters.
(See https://github.com/ros2/launch_ros/blob/d1c497878274a4ea6a1c1865fe74e395b961ba1f/launch_ros/launch_ros/utilities/normalize_parameters.py#L152 )

Resulting launch file:
```
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration, Command
from launch_ros.actions import Node

def generate_launch_description():
    conf_model = LaunchConfiguration('model')
    file_contents = Command(command=[
        "python3 -c '\n"
        "import yaml \n"
        "with open(\"", conf_model, "\") as f: \n"
        "    print(yaml.dump(f.read()))"
        "'"
    ])
    return LaunchDescription([
        Node(
            package="robot_state_publisher",
            name="robot_state_publisher",
            executable='robot_state_publisher',
            parameters=[{
                'robot_description': file_contents,
            }],
            output="screen",
        ),
    ])
```

Of course this is super hacky...

shonigmann, 2021-03-04:
@karl-schulz, in case you are still looking for a method that feels a bit less 'hacky', or in case others stumble upon this issue like I did, an option that is perhaps a bit cleaner would be to use xacro, as suggested [here](https://answers.ros.org/question/361623/ros2-robot_state_publisher-xacro-python-launch/). 

Code for reference:
```
#!/usr/bin/env python3

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration, Command
from launch_ros.actions import Node

def generate_launch_description():

    use_sim_time = LaunchConfiguration('use_sim_time', default='false')
    xacro_path = LaunchConfiguration('xacro_path', default=None)

    return LaunchDescription([
        DeclareLaunchArgument(
            'use_sim_time',
            default_value='false',
            description='Use simulation (Gazebo) clock if true'),
        DeclareLaunchArgument(
            'xacro_path',
            default_value=None,
            description='path to urdf.xacro file to publish'),

        Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            parameters=[{
                'use_sim_time': use_sim_time,
                'robot_description':Command(['xacro',' ', xacro_path])
            }]),
    ])
```

Generating urdf with xacro macros would seem to make sense to me for your case of having 2 similar robots, though you can easily convert your existing urdf to xacro by adding the xacro namespace to the <robot> tag at the head of your file. 
e.g. 
`<robot name="turtlebot3_waffle>` 
would become:
`<robot name="turtlebot3_waffle" xmlns:xacro="http://ros.org/wiki/xacro">`

karl-schulz, 2022-01-27:
Hi @shonigmann thanks for the tip. Unfortunately, I can't get it to work. Did you try it yourself?

Launch does create a temporary .yaml file in /tep when a launch file is executed. This yaml is passed to the nodes. Putting the full generated URDF string in there breaks the file and it cannot be parsed...

I get the following error: 
```
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 171, in parse_document_start
    raise ParserError(None, None, yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "<unicode string>", line 56, column 3:
      <link name="base_link">
```

This indicates that parts of the urdf are tried to be parsed as yaml during launch...

I will create a temporary file using xacro and provide it as a command line argument to the robot_state_publisher for now.

karl-schulz, 2022-01-27:
I found the reason: Whenever colons `:` or hash signs `#` are used in the output file (e.g. in a comment), the yaml generation produces an unparseable yaml. That's definitely a shortcoming of the current launch system...

shonigmann, 2022-01-27:
Hi @karl-schulz - yes, there is a known issue the xacro conversion into a parameter. There's more discussion in ros2/launch_ros#136, but it seems like its now possible to explicitly treat the param as a string to avoid this issue (e.g. `<param name="foo" value="$(command 'cat foo.urdf')" type="str"/>`)

karl-schulz, 2022-01-28:
Oh, I was not aware of this issue. Thanks for letting me know. Unfortunately there is apparently no workaround for python launch files...