ros/robot_state_publisher #156: Inconsistency in launch examples and ROS2 launch xml specifications

incebellipipo, 2021-02-08:
`<param/>` key no longer substitute `$(command ...)`. But it's still in use in the examples. 
_Branch is Foxy_
https://github.com/ros/robot_state_publisher/blob/d8b471b742bd16734184167622eca3c6dce10980/launch/rsp-launch-urdf-file.xml#L37

Am I missing something? If a change needed I can make it.

clalancette, 2021-02-09:
As far as I can tell, all of the examples work fine on Foxy.  Can you please give the error you are seeing when running it?

incebellipipo, 2021-02-09:
_Here is my intention if it helps. I'm trying to port UUV models into [uuv simulator](https://uuvsimulator.github.io/) ROS2 port [Plankton](https://github.com/Liquid-ai/Plankton)._ 

If I run the example it works like a charm.
```bash
 ~ ros2 launch robot_state_publisher rsp-launch-urdf-file.xml          
[INFO] [launch]: All log files can be found below /home/cem/.ros/log/2021-02-09-11-18-06-411110-cem-x1e-196727
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [196827]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] [INFO] [1612887486.486491405] [robot_state_publisher]: got segment base_link
```

```xml
<!-- Working. Gives a deprecation warning as expected -->
<launch>

  <arg name="namespace" default="eca_a9"/>

    <node
      name="robot_state_publisher"
      pkg="robot_state_publisher"
      exec="robot_state_publisher"
      output="screen"
      args="$(find-pkg-share eca_a9_description)/robots/generated/$(var namespace)/robot_description"
      >
    </node>

</launch>
```

However, If I try to use the method that used in the example in another launch file like as shown it fails poorly.

```xml
<!-- Not working -->
<launch>

  <arg name="namespace" default="eca_a9"/>

    <node
      name="robot_state_publisher"
      pkg="robot_state_publisher"
      exec="robot_state_publisher"
      output="screen"
      >
      <param name="robot_description" value="$(command 'cat $(find-pkg-share eca_a9_description)/robots/generated/$(var namespace)/robot_description')"/>
    </node>

</launch>
```
And this is how it fails
```
 ~ ros2 launch eca_a9_description state_publisher.launch
[INFO] [launch]: All log files can be found below /home/cem/.ros/log/2021-02-09-11-20-45-846331-cem-x1e-204484
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=ScannerError(None, None, 'mapping values are not allowed here', <yaml.error.Mark object at 0x7f9d34d6e3d0>)>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 417, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 377, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 145, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 70, in evaluate_parameter_dict
    yaml_evaluated_value = yaml.safe_load(evaluated_value)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 58, in compose_document
    self.get_event()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 118, in get_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 193, in parse_document_end
    token = self.peek_token()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 129, in peek_token
    self.fetch_more_tokens()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 320, column 16:
          <!-- todo: gazebo needs a collision volum ... 
                   ^
```

At this moment, I'm not quite sure if it's the gazebo complaining or the launch file failed to parse the argument.

clalancette, 2021-02-09:
Oh, I see.  OK, this is a problem with the launch infrastructure and more complicated pieces of XML.  I'm not entirely sure of the fix here.  I'm going to close this as a duplicate of https://github.com/ros2/launch_ros/issues/214 , which is the same issue.

closed 2021-02-09 by clalancette

HappySamuel, 2022-03-17:
Hi @clalancette 

Not sure if you still have this issue persists or not. But here's how i get it work.

```
<launch>
    <arg name="rviz" default="true"/>

    <!-- Robot State Publisher -->
    <node pkg="robot_state_publisher" exec="robot_state_publisher" name="robot_state_publisher" output="screen">
        <param name="robot_description" value="$(command 'xacro $(find-pkg-share aaaaaaaa_description)/urdf/aaaaaaaa.urdf.xacro')"/>
    </node>

    <!-- Joint State Publisher GUI -->
    <node pkg="joint_state_publisher_gui" exec="joint_state_publisher_gui" name="joint_state_publisher_gui"/>

    <!-- Joint State Publisher -->
    <node pkg="joint_state_publisher" exec="joint_state_publisher" name="joint_state_publisher"/>

    <!-- RVIZ -->
    <group if="$(var rviz)">
        <node pkg="rviz2" exec="rviz2" name="rviz2" args="-d $(find-pkg-share aaaaaaaa_description)/rviz/urdf.rviz"/>
    </group>

</launch>
```

Best,
Samuel