ros/robot_state_publisher #157: Static transforms published repeatedly (ROS 2)

jacobperron, 2021-02-08:
Using the ROS 2 version of robot_state_publisher, I expected a single message to be published when running the node, but instead I see the same message (with varying timestamp) published repeatedly. This is because the node is using a ROS timer to do this publishing.

This is different than the behavior from ROS 1, where we get a single latched message. I think it is better to change the behavior in ROS 2, then we avoid needless amounts of publisher traffic.

clalancette, 2021-02-08:
I don't disagree with your overall point that we could reduce traffic by using a latched message, but...

> This is different than the behavior from ROS 1, where we get a single latched message.

Is that really true?  Here is the ROS 1 code:

https://github.com/ros/robot_state_publisher/blob/ebe1927e243c3fe84f29c164b70c87678e8e9b74/src/joint_state_listener.cpp#L86

As far as I can tell, ROS 1 is also continually publishing fixed transforms.

Because of that, I suspect that a lot of downstream consumers are expecting the continual publishing (and not the latched message).  Thus, if we were to go the route of having it be a transient-local publication, we should probably make it optional (and non-default).

jacobperron, 2021-02-08:
> As far as I can tell, ROS 1 is also continually publishing fixed transforms.

Actually, it's not (I tested it). It's using the variable `use_tf_static_` to signal that it is a "one shot" timer (ie. the timer callback is only called once).

Also, the publisher is *already* transient local (according to my QoS introspection).

jacobperron, 2021-02-08:
It's possible downstream consumers are incorrectly subscribing without transient local and it just happens to be working for them because of the continuous publishing. However, if they are using the TF tooling we provide, there shouldn't be an issue.

jacobperron, 2021-02-08:
I've proposed a change in https://github.com/ros/robot_state_publisher/pull/158 to address this.

It seemed like a good opportunity to remove the `use_tf_static` parameter and ROS timer altogether. Alternatively, we could implement some logic to make the timer callback only happen once if `use_tf_static:=true`, but that seemed a little more complex. I opted for removing the deprecated parameter instead.

clalancette, 2021-02-08:
> Actually, it's not. Trying it out I only see one message. It's using the variable `use_tf_static_` to signal that it is a "one shot" timer (ie. the timer callback is only called once).

Oh right, makes sense.

> It's possible downstream consumers are incorrectly subscribing without transient local and it just happens to be working for them because of the continuous publishing. However, if they are using the TF tooling we provide, there shouldn't be an issue.

Yeah, anyone who is using `tf2_ros` to subscribe should be fine.  However, I'm pretty sure that there are users who subscribe "by hand"; this caught a few people out when we switched tf2_ros to transient-local in either Dashing or Eloquent (I can't remember which).

In any case, thanks for the PR, I'll continue the conversation over there.

jacobperron, 2021-08-31:
Resolved in https://github.com/ros/robot_state_publisher/pull/158

closed 2021-08-31 by jacobperron