ros/robot_state_publisher #167: [ros2-testing] Unable to locate package ros-rolling-robot-state-publisher

vatanaksoytezer, 2021-06-28:
Today after the merge of https://github.com/ros/rosdistro/pull/30045 my colleague @JafarAbdi and I started to experience Github Actions failures in ros-planning repos with this error on CI runs that use rolling-testing repositories. See https://github.com/vatanaksoytezer/moveit2/runs/2934770279?check_suite_focus=true for an example run. rolling-main repositories works and passes CI perfectly fine.

I can also reproduce the issue if I change my ros repositories to testing ones in my local rolling setup.

I appreciate any insights, maybe @clalancette?

Thanks!

clalancette, 2021-06-28:
There was a slight hiccup there for a while when we had an infrastructure failure, and robot-state-publisher (along with a bunch of other packages) wasn't available.  It should be fixed now (it works here for me).

closed 2021-06-28 by clalancette

vatanaksoytezer, 2021-06-28:
I can confirm that the issue is fixed, thanks for fixing it quick!