ros/robot_state_publisher #181: Static fixed transforms are not updated with robot_description parameter

RussellJoyce, 2021-11-11:
In Galactic and later, when the robot URDF is updated at runtime by setting the `robot_description` parameter, static fixed transforms are currently not re-published to match the new model.

A workaround in Galactic is to use `tf_static := false`, but this obviously isn't ideal as the parameter has been deprecated, and it also has the side-effect of publishing static transforms 500ms into the future, still resulting in half a second of broken transform data after the URDF update.

A simple fix would be to call `publishFixedTransforms()` just after the URDF is updated in `parameterUpdate()` [here](https://github.com/ros/robot_state_publisher/blob/8b3f6b8e03432651e2b788128e17603f0972c677/src/robot_state_publisher.cpp#L357), to keep the transforms in line with the new URDF. The same fix should also work on Galactic, but should probably be predicated on `if (use_tf_static_)` to match the same behaviour in the initialiser.

I'm happy to put this into a PR, if this behaviour is indeed a bug.

clalancette, 2021-11-11:
Oh, excellent point.  That is indeed a bug.  I'd be happy to review a pull request to fix this; please make sure to target the `ros2` branch with the PR.

clalancette, 2021-11-12:
Fixed by #182, so closing.

PR #182 (2021-11-12): Publish new fixed transforms when URDF is updated
A new URDF may come with a new set of fixed transforms, so if the URDF
is updated through the `robot_description` parameter, we also need to
publish its new transforms.

Fixes #181.
commit 17af0ac5 (2021-11-12): Publish new fixed transforms when URDF is updated

A new URDF may come with a new set of fixed transforms, so if the URDF
is updated through the `robot_description` parameter, we also need to
publish its new transforms.

Fixes #181
closed 2021-11-12 by clalancette