ros/robot_state_publisher #187: Missing urdf_parser_plugin exec_dependencies?

RiZom-91, 2022-01-18:
Hi,

I've noticed this package depends on urdf for build and execution but urdf only depends on urdf_parser_plugin for build not execution. As a result, running ros2 run robot_state_publisher robot_state_publisher <urdf_file> fails with the following error on my setup:

clalancette, 2022-02-07:
Sorry for the long delay in answering here.  This error is pretty weird to me, because `urdf_parser_plugin` isn't an instantiated class; it is only a header file.  So it really should only be necessary at build time, not at runtime.  Can you give more details on what ROS 2 distribution you are using, what URDF file you are using, and the version of `robot_state_publisher` and `urdf_parser_plugin` you have?

RichProgrammerEE, 2022-07-11:
@RiZom-91 Could you please elaborate on what you did to work around this issue. I am encountering the same issue and adding `ROS_EXEC_DEPENDS = "urdf-parser-plugin"` to my bitbake recipe file does not seem to fix the issue? Any advice would be appreciated.

whni, 2024-05-05:
@clalancette Hi I am encountering the same issue. Not sure if this issue is still under track. My env is Yocto (kirkstone) + ROS 2 (humble)
```
[robot_state_publisher-1] terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
[robot_state_publisher-1]   what():  package 'urdf_parser_plugin' not found, searching: [/usr]
[ERROR] [robot_state_publisher-1]: process has died [pid 933, exit code -6, cmd '/usr/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_vcvbvyxx --params-file /tmp/launch_params_kvj4v1d7'].
```

whni, 2024-05-06:
Solved by adding 
```
RDEPENDS:${PN} += "urdf-parser-plugin"
```
to urdf_%.bbappend

[image]