ros/robot_state_publisher #192: frame_prefix parameter should change to tf_prefix parameter

MortezaAliyari, 2022-03-06:
hi,
when I use below code:
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher" output="screen">
      <param name="publish_frequency" type="double" value="50.0" />
      <param name="tf_prefix" value="tb3_1_tf" />
    </node>
The tf tree of the robots is fine in contrast to using the frame_prefix parameter.
The ROS version is Noetic and rviz version is 1.14.14, all executed in Ubuntu focal.

clalancette, 2022-03-06:
Sorry, I'm not sure what you mean.  In ROS 1 noetic, using `tf_prefix` is the correct way to add prefixes to TF.  `frame_prefix` is new in ROS 2, so doesn't apply to ROS 1.  Can you explain more about what you are expecting to happen?

MortezaAliyari, 2022-03-10:
I was trying to apply 'frame_prefix' to ROS1.
Sorry for my mistake.