ros/robot_state_publisher #195: Missing parameter `source_list` when porting from ROS 1 to ROS 2

adeguet1, 2022-04-04:
In ROS1, the `robot_state_publisher` had an option to subscribe to multiple topics to retrieve the joint state (`source_list`).  This can be used with a device (e.g. gripper) attached to another robot (e.g. UR).  Then both the gripper and UR controllers push their own joint state.  Since the state publisher looks for names in the `joint_state`, it can pick the joint values from one or more topics.  This worked as long as the joint names are unique, including across topics.

Is this supported in ROS 2 and I missed it?  If not, is there an alternative approach in ROS?  If the answer is still no, is this a feature that could be added?   If so, I could try to implement this and create a PR.

sloretz, 2022-04-04:
I don't see a `source_list` parameter in `robot_state_publisher` in ROS 1 or ROS 2, but [I do see one in `joint_state_publisher`](https://github.com/ros/joint_state_publisher/blob/af3dd1bec348adba7879e3e7a866c5a28816bd38/joint_state_publisher/README.md?plain=1#L31). Is that what you're thinking of? If so, [it's also available in ROS 2](https://github.com/ros/joint_state_publisher/blob/e7bb91b05dfa6a704a7a2cad2650b4c97a4b93ad/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L223).

I expect remapping both the gripper's and controller's joint states to the same topic and having `robot_state_publisher` subscribe to that would work as well.

adeguet1, 2022-04-11:
# Solution 1:

Using the `joint_state_publisher`, it works but I'm getting a slightly confusing message:
```
Waiting for robot_description to be published on the robot_description topic...
```
It would be nice to get a confirmation message once the `joint_state_publisher` finally gets the `robot_description`.  Alternatively, maybe the `joint_state_publisher` could have an option to use a node/parameter for the `robot_description`.

# Solution 2:

Does the topic remapping happen on the publisher's node or on the subscriber node?  I tried to modify the launch for the `robot_state_publisher` using something like:
```python
    robot_state_publisher_node = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        name='robot_state_publisher',
        remappings=[
            ('joint_states', 'PSM1/jaw/measured_js'),
            ('joint_states', 'PSM1/measured_js')],
        parameters=[{'use_sim_time': use_sim_time,
                               'robot_description': ParameterValue(
                                    Command(['xacro ', str(urdf)]), value_type=str),
        }],
        arguments=[urdf]
    )
```
But that didn't work.  I assume the remapping is some kind of map of subscriptions, not a multimap?  So the first element of the remapping can't be repeated...