ros/robot_state_publisher #198: [ROS2] robot_state_publisher start failure

chenjunnn, 2022-05-14:
Using docker image ros:galactic-ros-base
```
[robot_state_publisher-1] Error:   Error document empty.
[robot_state_publisher-1]          at line 71 in /tmp/binarydeb/ros-galactic-urdfdom-2.3.5/urdf_parser/src/model.cpp
[robot_state_publisher-1] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[robot_state_publisher-1] terminate called after throwing an instance of 'std::runtime_error'
[robot_state_publisher-1]   what():  Unable to initialize urdf::model from robot description
[ERROR] [robot_state_publisher-1]: process has died [pid 6240, exit code -6, cmd '/opt/ros/galactic/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_w4reobq_'].
```

chenjunnn, 2022-05-14:
Sorry, I found that it was caused by my using urdf path directly as the `robot_description` param.

closed 2022-05-14 by chenjunnn

Vinothhk, 2023-11-26:
can you share the launch file code pls

Vinothhk, 2023-11-26:
I am encountering the error. But couldn't solve