ros/robot_state_publisher #199: Python Bindings for Robot State Publisher

geoeo, 2022-09-07:
Would it be possible to write python bindings for the accessing the library?

I think currently its only possible to manually publish a robot state publisher via cpp code?

clalancette, 2022-09-07:
It would be possible, but I'm not entirely sure why you would want to do it.  As it stands, `robot_state_publisher` does exactly one thing; it subscribes to `/joint_states`, and given a URDF, it publishes the joint states out to `/tf` or `/tf_static`.  There isn't much of an API to call here.

Can you explain why you want a Python API, and what you are trying to accomplish?

geoeo, 2022-09-07:
I have a mesh of an object im tracking that I want to visualize in rviz. Since I can only publish the robot state model via the node I either 1) need an extra terminal or 2) use the launch file. When Im developing I dont want to use the launch file. So I need an extra terminal. It would be more ergonomic (in my opinion) if I could publish all data from my node (via code). I guess this is more a quality of life improvement