ros/robot_state_publisher #205: frame_prefix parameter not working

hcdiekmann, 2023-03-08:
ROS 2 Foxy
Ubuntu 20.04
In an attempt to launch multiple RSP nodes in different namespaces the frame_prefix parameter is not remapping the `/tf` topic.
```python
  launch_ros.actions.Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            output='screen',
            namespace=launch.substitutions.LaunchConfiguration('robot_namespace'),
            parameters=[{
                'robot_description': launch.substitutions.LaunchConfiguration('urdf'),
                'frame_prefix': launch.substitutions.LaunchConfiguration('robot_namespace'),
                'use_sim_time': launch.substitutions.LaunchConfiguration('use_sim_time'), }])
```
Interestingly, the robot description however is being published in the correct namespace. As seen in the node/topic graph below.

[image]

clalancette, 2023-03-08:
> In an attempt to launch multiple RSP nodes in different namespaces the frame_prefix parameter is not remapping the `/tf` topic.

`frame_prefix` is not responsible for remapping the `/tf` topic name.  What it does is to add in a prefix to the frame_id that is being published as part of the TransformStamped message fields `header.frame_id` and `child_frame_id`.  If you also want to remap `/tf`, you can do that using the normal remapping rules.

hcdiekmann, 2023-03-08:
That makes sense. 
By normal remapping rules, you refer to these [ROS name remapping ](https://design.ros2.org/articles/static_remapping.html)?

clalancette, 2023-03-08:
> By normal remapping rules, you refer to these [ROS name remapping ](https://design.ros2.org/articles/static_remapping.html)?

Yes, exactly.

With that, I'm going to close this out.  But feel free to keep commenting if you need.

closed 2023-03-08 by clalancette