ros/robot_state_publisher #216: Have the robot_state_publisher ignore certain joints

firesurfer, 2024-02-26:
Hi, we have a fairly complex setup where calibration is rather difficult. In order to directly see the result of our calibration efforts we would like to manually publish the transforms between certain links (e.g. the connection of the robot arm base with its mounting). Is it possible to tell the `robot_state_publisher` to not publish transforms for certain joints?

sloretz, 2024-02-26:
> In order to directly see the result of our calibration efforts we would like to manually publish the transforms between certain links (e.g. the connection of the robot arm base with its mounting). Is it possible to tell the robot_state_publisher to not publish transforms for certain joints?

For any joint that you're going to manually publish the transform I'd recommend [making its type be `"floating"` joint in your URDF](https://wiki.ros.org/urdf/XML/joint#Attributes). `robot_state_publisher` won't try to publish the transforms because there's no joint state message that could describe 6 dof movement, and you're custom transform publisher can publish any transform it wants.

https://github.com/ros/robot_state_publisher/blob/e0f77e62457ae01f8dbf3b5c4cc62d15fa36dfad/src/robot_state_publisher.cpp#L230-L236

closed 2024-02-26 by sloretz

firesurfer, 2024-02-27:
Thanks a lot. That is exactly what I was looking for!