ros/robot_state_publisher #219: Intra-process comm not supported due to `transient_local` durability

roncapat, 2024-06-26:
Currently, `robot_state_publisher` can be used as a composable node, but with Intra-Process disabled. It would be desiderable to allow its enablement.

sloretz, 2024-06-26:
Mind trying if intra-process works when you [override the qos parameter](https://design.ros2.org/articles/qos_configurability.html)?

The parameter name should be something like: `qos_overrides./robot_description.publisher.durability` and you can set it to `volatile`.

Yadunund, 2024-06-26:
@sloretz's solution would work but often times downstream subscriptions expect a `transient_local` durability. Hence, I've opened https://github.com/ros/robot_state_publisher/pull/220. Could you give that a try?

roncapat, 2025-07-11:
I'm still having difficulties in Jazzy with this one. IIRC we assumed that IPC `transient_local` support in Jazzy would solve this one (hence the PR closed), but I'm re-trying now and it seems not to work.