ros/robot_state_publisher #227: frame_prefix not being applied in humble

agillies8, 2024-11-21:
Ros2 Humble
In a docker container that pulls from:
```
ARG ROS_DISTRO=humble

########################################
# Base Image for ISAAC CAMERA EXTENSION #
########################################
FROM osrf/ros:${ROS_DISTRO}-desktop AS base
ENV ROS_DISTRO=${ROS_DISTRO}
```

Im attempting to add a prefix to the tf names like so, since I have multiple robots Im trying to get all on the same /tf message since they interact. Im doing it in the launch file as so:

```
        Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            namespace='kicker_01',
            parameters=[{
                'use_sim_time': use_sim_time,
                'robot_description': robot_desc,
                'frame_prefix': "World_Kickers_side_kicker_01_",
                'publish_frequency': 10.0,}],
            ),
```

However when I bring up rviz, the robot model is still using the native tf frame names - its like the frame_prefix param is not being applied. Any thoughts on how to troubleshoot?

(also as an additional note - Im pretty sure the robot model display type in rviz2 is out of date, since it still shows the old 'tf_prefix' param from ros1, and you you add in a prefix there, it adds a "/" to your prefix name.

agillies8, 2024-11-21:
I should also add, I can get the param while the node is running and it looks like it was set on the parameter server:

```
$ ros2 param get /kicker_01/robot_state_publisher frame_prefix
String value is: World_Kickers_side_kicker_01_
```

agillies8, 2024-11-22:
Ok update - I now realize that it was actually correctly appending the 'frame_prefix' as advertised, and this can be seen using the ros2 run tf2_tools view_frames tool.

What was tripping me up was when I added the "Robot Model" display in Rviz, the tf_prefix setting was not populating when you subscribe to the robot description. For some reason I assumed that when you selected the robot description message, that would carry with it any TF mappings or prefixes. Now I understand it doesn't and you have to adjust that in Rviz separately. 

I still think thats a bug in Rviz, that it should also be renamed to 'frame_prefix', and also not force you to append a '/' after your prefix, to follow the style in robot model.

Nonetheless, I think I've gone full circle on this now.

Raivias, 2025-04-04:
Came across this because I had the same prefix issue. I think the '/' is forced because situationally you might use a "_" style prefix depending on the situation. But that seems like it's on the rviz side.

Can this issue be closed?